amaranth-lang/amaranth #318: pysim: Throw Error or Warning on writing to non-port variable

mszep, 2020-02-10:
As a beginner, this tripped me up:

I was changing a Signal's value both in the module's internal logic and in my sync process during simulation, and wondering why I was getting nondeterministic results.

If there are legitimate reasons to drive internal (non-port) Signals from a test bench, perhaps throwing a Warning is sufficient, or allowing an optional parameter `allow_drive_internal_signal=True` passed to the simulator constructor?

whitequark, 2020-02-10:
This is indeed a serious issue that I was considering as a part of #228, but it's important enough to have its own bug too. What you suggest would work, but I have in mind a plan for a more flexible approach; I haven't yet described it in detail though.

whitequark, 2024-06-11:
In the 0.5 release this will raise a `DriverConflict` error with the message "Combinationally driven signals cannot be overriden by testbenches".

closed 2024-06-11 by whitequark