amaranth-lang/amaranth #324: Integrate cxxrtl simulator

whitequark, 2020-02-16:
A while ago I've implemented the [Yosys cxxrtl backend](https://github.com/YosysHQ/yosys/pull/1562), which is very similar to the current Python simulator conceptually while being significantly (10-100×) faster, competitive with Verilator.

nMigen should tightly integrate with this simulator, providing an interface identical to that of `nmigen.back.pysim`.

cestrauss, 2020-11-06:
As the cxxsim work nears completion, I would like to point out some remaining issues, so they are not missed:

- In the VCD file generated by cxxsim, there is no ``top`` module, like in pysim.
This means the same GTKWave document can't be used for both.

- Enum traces are not supported (the numerical values are recorded instead).

- There are still a couple of FIXMEs in write_vcd (write the gtkw_file and use the traces parameter).

whitequark, 2020-12-03:
CXXSim no longer ignores contract violations, and uses an appropriate method to reset the simulation.

Prepending the `top` module in CXXSim VCD files will require upstream changes, see new task in #531.

whitequark, 2020-12-03:
VCD hierarchy in CXXSim and PySim now matches.

whitequark, 2020-12-05:
CXXSim now writes GTKW files.

whitequark, 2020-12-08:
CXXSim will not support race-free memory read/write operations; see https://github.com/nmigen/nmigen/issues/531#issuecomment-740512280 for details.

robtaylor, 2022-08-16:
I'm happy to have a look at it on mac when time is right..