amaranth-lang/amaranth #325: pysim TypeError: unsupported operand type(s) for &: 'tuple' and 'int'

awygle, 2020-02-17:
After trying to replace some `nmigen.compat.genlib.Record` usages with `nmigen.hdl.rec.Record`, I'm getting this error when attempting to run the `test_analyzer` test case on my litescope port, as can be seen here: https://github.com/awygle/darkscope/tree/96eaaf9e72a2665e89e69237db42246c0a40bd79

Sorry for the terribleness of this bug report, but I couldn't manage to reduce the test case in any meaningful way....

awygle, 2020-02-18:
Hooray, I reduced it to a test case that fits in an issue comment!

```python
import unittest

from nmigen import *
from nmigen.hdl.rec import Record, DIR_FANOUT, DIR_FANIN, DIR_NONE, Layout
from nmigen.back.pysim import *

class Repro(unittest.TestCase):
    def test_repro(self):
        dut = Module()
        full_layout = [
            ("valid",   1, DIR_FANOUT),
            ("ready",   1, DIR_FANIN),
        ]
        src = Record(full_layout, name="testsrc")
        sink = Signal(src.shape().width)
        dut.d.sync += sink.eq(src)

        sim = Simulator(dut)
        sim.add_clock(1e-6, domain="sync")
        def process():
            yield Tick()
        sim.add_process(process)
        sim.run()
```

awygle, 2020-02-18:
Okay, I messed it up _again_. The problem here is the empty subrecord:

```python

import unittest

from nmigen import *
from nmigen.hdl.rec import Record, DIR_FANOUT, DIR_FANIN, DIR_NONE, Layout
from nmigen.back.pysim import *

class Repro(unittest.TestCase):
    def test_repro(self):
        dut = Module()
        full_layout = [
            ("valid",   1, DIR_FANOUT),
            ("ready",   1, DIR_FANIN),
            ("payload", [("data", 1, DIR_FANOUT), ("hit", 1, DIR_FANOUT)]),
            ("param", []),
        ]
        src = Record(full_layout, name="testsrc")
        sink = Signal(src.shape().width)
        dut.d.sync += sink.eq(src)

        sim = Simulator(dut)
        sim.add_clock(1e-6, domain="sync")
        def process():
            yield Tick()
        sim.add_process(process)
        sim.run()

```

If you remove the empty subrecord, the test passes.

whitequark, 2020-02-19:
Minimized:
```python
from nmigen import *
from nmigen.back.pysim import *

dut = Module()
dut.d.comb += Cat(Signal()).eq(Cat())
Simulator(dut).run()
```

commit ec7aee62 (2020-02-19): back.pysim: fix RHS codegen for Cat() and Repl(..., 0).

Fixes #325.
closed 2020-02-19 by whitequark