amaranth-lang/amaranth #327: pysim: option to capture simulation output in a python object

mszep, 2020-02-28:
Would it make sense to have a way to capture simulation output in a python object, rather than having to use a vcd file and a separate reader program?

I'm building a physics simulator in nmigen and it would be super helpful when debugging to be able to plot a signal's trace (over eg 1000 timesteps) in matplotlib, within the same python session as the simulation.

Does this make sense as a workflow?
Is there already a way to do this?

whitequark, 2020-02-28:
Yes, it would make perfect sense as a workflow. Do you have any preferences regarding an exported API here?

mszep, 2020-02-28:
Thinking about the API, I'd have no preference -- whatever's most convenient on the nmigen side.

I guess for consistency we could have a context manager similar to `write_vcd`, so something like

```
traces = {}
with write_dict(traces):
    sim.run()
```

awygle, 2020-02-28:
I'd like us to make sure that this can be used incrementally - that is, run for N clock ticks and return the traces, do something with them, then run for M more clock ticks and return those traces. This ties into my dream of an interactive simulator.

Note that I don't think the context manager API is incompatible with the above, just making a use case known. `run_until` and `step` in place of `run` should cover it.

mszep, 2020-02-29:
Yes, that would be awesome! I was thinking of the possibility of making interactive simulations with ipywidgets, but if making it iterative will complicate the API or the implementation of the simple case, perhaps the simple case should take priority.

whitequark, 2020-02-29:
It wouldn't be hard to make it iterative; the VCD writer isn't, but only due to limitations of the VCD file format.

mszep, 2020-03-07:
FWIW, I ended up just passing a StringIO object to the `write_vcd` context manager and parsing the vcd text data into plottable form.

Therefore, I think this issue can be deprioritized or closed altogether.

whitequark, 2020-03-08:
I think it would be an appropriate and highly useful feature to have: your workaround is... let's say suboptimal (no shade but it's pretty inefficient), and what @awygle wants here is also very reasonable.

alanvgreen, 2021-08-20:
This looks like a typical application for an Observer. The Observer is registered with the simulator, which then notifies it of interesting events.

I made a draft PR as an RFC. I think this is general enough to:

- act as the interface for _VCDWriter
- capture dictionaries of values at each step for @mszep's use case
- intercept simulator state at defined numbers of clock cycles or timestamps for @awygle's use case

Please send comments.

whitequark, 2023-02-03:
@alanvgreen https://github.com/amaranth-lang/amaranth/pull/628#issuecomment-1414802510