amaranth-lang/amaranth #328: Add (optional) support for watermarks in FIFOs

awygle, 2020-03-05:
It's often useful to have a "high watermark/low watermark" signal from a FIFO, particularly when processing any kind of fixed-length frames. This could be implemented as a wrapper around `SyncFIFO` pretty easily but probably wants to be more deeply integrated with `AsyncFIFO` since there's already a clock-domain-crossing counter to use (which is not currently exposed to consumers).

whitequark, 2020-03-05:
I'm not sure how an async FIFO with water marks works, CDC-wise. Do you have any references?

awygle, 2020-03-05:
As section 5.6 of http://www.sunburst-design.com/papers/CummingsSNUG2002SJ_FIFO1.pdf says, you compare against (ptr+N) instead of (ptr) as you would for empty/full. That section appears to be discussing FIFO style 1, but the same technique should be adaptable to FIFO style 2 with minor modifications.

whitequark, 2020-03-05:
That means we have to Gray decode the resynchronized counters, right? Then it seems like we should rather have `r_level` and `w_level` for consistency with `SyncFIFO`, and the consumer of the FIFO interface can then write a comparator with HWM/LWM. We can even make it a part of the general FIFO interface, with the SyncFIFO case aliasing `level` as both.

awygle, 2020-03-05:
That makes sense, then we can always create a HWM/LWM wrapper as a utility module if we want to and it can work with any `FIFOInterface` instance.

whitequark, 2020-03-05:
In particular I think having a level is strictly more useful than watermarks (even configurable watermarks), if only for debugging. It's true that comparators are cheaper on architectures which have lookahead carry primitives, but I would say that it's the toolchain's job to simplify subtract-then-compare, since that's a fairly simple local transformation.

awygle, 2020-03-05:
I'll be honest, I missed `level` on SyncFIFO when I was looking at the existing FIFO code 😅

whitequark, 2020-11-03:
Fixed in https://github.com/nmigen/nmigen/pull/472, https://github.com/nmigen/nmigen/pull/512, https://github.com/nmigen/nmigen/pull/513, https://github.com/nmigen/nmigen/pull/524.

closed 2020-11-03 by whitequark