amaranth-lang/amaranth #331: Probable submodule name collision breaking modularity

galibert, 2020-03-05:
I may have missed something, but...

Imagine you create a module of type M68000.  It has a submodule of type BusAccess.  To get a meaningful name in the rtlil and every tool that uses it, it is created in the recommended way in elaborate:
- m.submodules.busaccess = busaccess = BusAccess()

Then the name "busaccess" is used, which is good.  But it seems that module names must be unique in the whole system.  So does that mean one couldn't have two instances of M68000 in the same system if we want significant names?

whitequark, 2020-03-05:
It doesn't need to be unique in the entire design; it only needs to be unique within a single parent module.

galibert, 2020-03-05:
Ok, I can confirm nmigen automatically renames the colliding names (with a $number suffix) to avoid collisions, very nice.

closed 2020-03-05 by galibert