amaranth-lang/amaranth #339: Add rotate left/right

ZirconiumX, 2020-03-23:
While it's not hard to roll your own in a function, it seems a bit more ergonomic to have `Value.rotate_{left,right}()` instead.

whitequark, 2020-03-23:
Variable rotates or constant rotates?

ZirconiumX, 2020-03-23:
Currently I only need constant rotates, but since you can build a rotate out of shifts, it shouldn't be difficult to also handle variable rotates.

whitequark, 2020-03-23:
Constant rotates are easy and uncontroversial. Variable rotates not so much because if the LHS is not a power of 2 and the RHS can represent an absolute value larger than the length of LHS, you get a (possibly signed) modulus operator in the expansion and that's likely going to result in a nasty synthesized netlist.

ZirconiumX, 2020-03-23:
Excellent point, I'd forgotten about that (and this is why I filed an issue first). I suppose it would have to be constant right hand side then.

whitequark, 2020-03-23:
We can always extend it to varshifts later if someone comes up with a good way to handle this issue. So let's stick to constshifts for now.

PR #352 (2020-04-13): Add rotate by constant
I think I got the signs right, anyway.

Closes #339.
closed 2020-04-13 by whitequark