amaranth-lang/amaranth #340: Accessing Memory objects with '[ ]' array syntax causes confusing build errors.

WRansohoff, 2020-03-27:
It sounds like it is only possible to access Memory data with square-bracket array syntax (`memory[ index ]`) during simulation.

But when I try to build a bitstream from a design which uses that syntax, it seems to cause an infinite loop in nMigen instead of printing an error. Interrupting the process results in some nested `LegalizeValue` exceptions which eventually point to:

    nmigen.back.rtlil.LegalizeValue: ((+ (sig addr) (const 2'd2)), range(0, 1028), ('[...]/nmigen/hdl/mem.py', 91))

It sounds like this can be fixed by using the Memory object's read/write ports [like the `examples/basic/mem.py` file](https://github.com/nmigen/nmigen/blob/master/examples/basic/mem.py) demonstrates, but an error message might help make that more discoverable. Thanks!

FFY00, 2020-03-27:
Hum, shouldn't `RecursionError` be raised? I think it would be good to investigate why it isn't triggered.

whitequark, 2020-03-28:
> Hum, shouldn't `RecursionError` be raised? I think it would be good to investigate why it isn't triggered.

Because it's not an infinite loop or recursion. It's just really slow.

whitequark, 2024-02-01:
https://github.com/amaranth-lang/rfcs/pull/45 may address this issue if accepted.

closed 2024-02-09 by whitequark