amaranth-lang/amaranth #343: Provide a way to add a reset Signal() to default clock domains

WRansohoff, 2020-03-29:
It would be nice to be able to add an internal reset signal to a platform's default clock domains. It looks like you can set `default_rst` in a board file, but it also looks like that needs to be a platform Resource such as an I/O pin.

In a nutshell, it'd be great if this worked (assuming that the `elaborate` method also contains some logic that uses `m.d.sync` so it doesn't get optimized away):

```
class SomeDesign(Elaboratable):
  def __init__(self):
    self.reset_sig = Signal(1, reset=0b0, reset_less=True)

  def elaborate(self, platform):
    m = Module()
    m.d.comb += m.d.sync.rst.eq(self.reset_sig)
    return m
```

But that results in an `AttributeError: '_ModuleBuilderDomain' object has no attribute 'rst'`. It looks like there is also a `ResetSignal()` function, but I couldn't get that to work either. Maybe I was calling it wrong, the closest I got was:

```
[...]
def elaborate(self, platform):
  m = Module()
  m.d.comb += ResetSignal(domain="sync").eq(self.clk_rst)
  return m
```

That returns an AssertionError with the top level of the stack trace [pointing here](https://github.com/nmigen/nmigen/blob/2d1e12d00ca13ce6ad92fe145beeaa74925de6ef/nmigen/hdl/ir.py#L396).

It looks like it is possible to re-create the 'sync' domain using the same logic as `create_missing_domain`, but [I had to make a small change to that method to accept an arbitrary reset Signal](https://github.com/WRansohoff/nmigen/commit/0efa93ba21284278275f7dc2132317b94294002b) for this to work:

```
[...]
def elaborate(self, platform):
  m = Module()
  if platform != None:
    platform.create_missing_domain("sync", reset_sig=self.clk_rst)
  return m
```

Is there maybe a way to access the global 'sync' instead of trying to modify it from `m.domains`? Or some other way to do this that I'm not thinking of? Thanks!

whitequark, 2020-03-29:
As discussed on IRC, in this case, the easiest and really the most natural way would be to use a `ResetInserter`. Specifically, wrap all of your code that requires a reset in a module, say `your_logic`, and then add `ResetInserter(your_reset)(your_logic)` in your toplevel module rather than `your_logic` directly.

WRansohoff, 2020-03-29:
Great, thanks!

closed 2020-03-29 by WRansohoff