amaranth-lang/amaranth #345: Simulation of module with undriven signal causes generated VCD to not contain the signal, but contain an individual signal for each character of the undriven signal's name

The6P4C, 2020-04-01:
This snippet illustrates the issue:

```py
from nmigen import *
from nmigen.back import pysim

class Bug(Elaboratable):
    def __init__(self):
        self.reg = Signal()
        self.out = Signal()

    def elaborate(self, platform):
        m = Module()
        m.d.sync += self.out.eq(~self.out)

        # driving the signal fixes the issue
        # uncomment: vcd will be correct
        # m.d.comb += self.reg.eq(0)

        return m

bug = Bug()
fragment = Fragment.get(bug, None)
sim = pysim.Simulator(fragment)
sim.add_clock(1e-6)
with sim.write_vcd(vcd_file='bug.vcd', traces=[bug.out, bug.reg]):
    sim.run_until(1e-6 * 10, run_passive=True)
```

With the line driving `reg` inside `elaborate` commented out, the generated VCD starts:
```
$comment Generated by nMigen $end
$date 2020-04-01 22:07:22.464464 $end
$timescale 100 ps $end
$var wire 1 3 r $end
$var wire 1 4 e $end
$var wire 1 5 g $end
$scope module top $end
$var wire 1 0 clk $end
$var wire 1 1 rst $end
$var wire 1 2 out $end
$upscope $end
$enddefinitions $end
```

i.e. `reg` has somehow become three signals, `r`, `e` and `g`.

Driving `reg` combinatorically or with registered logic fixes the signal's name and value in the VCD.

The6P4C, 2020-04-01:
Experimenting, it appears this happens if the signal is unused in any way - either undriven, or not part of the module's logic.

whitequark, 2020-04-01:
Ouch, I think I made a mistake fixing #280.

commit ec8386a7 (2020-04-03): back.pysim: fix emission of undriven traces to VCD files.

This has been originally implemented in commit d3775eed (which fixed
`write_vcd(traces=)` to do something at all), but had a flaw where
undriven traces would not be correctly placed in hierarchy. This
used to produce incorrect results on pyvcd 0.1, but started causing
assertion failures on pyvcd 0.2.

Fixes #345.
closed 2020-04-03 by whitequark