amaranth-lang/amaranth #347: DriverConflict raised on entire signal when independently driving individual bits from submodules

The6P4C, 2020-04-05:
Hopefully this isn't just me doing something wrong, but, the following snippet

```py
from nmigen import *

class MySubmodule(Elaboratable):
    def __init__(self, o):
        self.o = o

    def elaborate(self, platform):
        m = Module()
        m.d.comb += self.o.eq(1)
        return m

class Bug(Elaboratable):
    def elaborate(self, platform):
        m = Module()

        xo = Signal(2)
        m.submodules += [
            MySubmodule(xo[0]),
            MySubmodule(xo[1])
        ]

        return m

from nmigen.back import verilog
verilog.convert(Fragment.get(Bug(), None))
```

produces the error

```bug.py:16: DriverConflict: Signal '(sig xo)' is driven from multiple fragments: top.<unnamed #0>, top.<unnamed #1>; hierarchy will be flattened```

despite each individual bit of `xo` only being driven in one location.

whitequark, 2020-04-06:
This is an unavoidable limitation caused by the fact that module ports have per-port direction (in RTLIL, Verilog, etc), but your code would only work if they had per-bit direction. See also #191.

whitequark, 2020-04-06:
Someone on IRC (mwk?) suggested that this could be worked around using inout ports, but this would cause all sorts of issues downstream.

closed 2020-04-06 by whitequark