amaranth-lang/amaranth #359: Python chokes on simulated memories of depth greater than ~3k

sjolsen, 2020-04-17:
With Python 3.7.3 (shipped with Debian stable) and 3.8.2 (via pyenv), I encountered the following error when simulating a design with a couple of memory banks:

```
  File "/path/to/nmigen/back/pysim.py", line 765, in __call__
    exec(emitter.flush(), exec_locals)
RecursionError: maximum recursion depth exceeded during compilation
```

Here's a minimal reproduction:

```python
import unittest

from nmigen import *
from nmigen.back.pysim import *

class MemTest(unittest.TestCase):

    def test_mem(self):
        m = Module()
        mem = Memory(width=1, depth=0xC00)
        m.submodules.rport = mem.read_port()
        m.submodules.wport = mem.write_port()
        sim = Simulator(m)
        sim.run()
```

Reducing the depth of the memory (width doesn't seem to matter) makes the failure go away. For the time being that workaround works for me, but it would be nice to use the same design parameters for simulation and synthesis.

whitequark, 2020-04-17:
Yep, this is a known pysim issue that requires reworking the way memories are translated.

whitequark, 2020-04-30:
Interesting, I thought this would work on 3.7 (https://bugs.python.org/issue31113) but it doesn't.

twam, 2021-04-12:
Observed the same problem. Can workaround for now by calling `sys.setrecursionlimit(x)`.

jeremyherbert, 2022-01-02:
Just to add a few notes to this to help with people finding it from google:

On a Core i9 MBP (2019) this makes simulations of memory with a depth of 2**13 (just a lookup table) take about 1 minute if the recursion limit is set high enough. Python version is `3.9.5 (v3.9.5:0a7dcbdb13, May  3 2021, 13:05:53) \n[Clang 12.0.5 (clang-1205.0.22.9)]`.

You can expect to see an exception which looks something like this:

```
Traceback (most recent call last):
  File "/Users/jeremy/Seafile/misc_projects/amaranth_playing/sine_lookup.py", line 96, in <module>
    sim = Simulator(m)
  File "/Users/jeremy/Seafile/misc_projects/amaranth_playing/venv/lib/python3.9/site-packages/amaranth/sim/core.py", line 68, in __init__
    self._engine   = engine(self._fragment)
  File "/Users/jeremy/Seafile/misc_projects/amaranth_playing/venv/lib/python3.9/site-packages/amaranth/sim/pysim.py", line 282, in __init__
    self._processes = _FragmentCompiler(self._state)(self._fragment)
  File "/Users/jeremy/Seafile/misc_projects/amaranth_playing/venv/lib/python3.9/site-packages/amaranth/sim/_pyrtl.py", line 463, in __call__
    processes.update(self(subfragment))
  File "/Users/jeremy/Seafile/misc_projects/amaranth_playing/venv/lib/python3.9/site-packages/amaranth/sim/_pyrtl.py", line 463, in __call__
    processes.update(self(subfragment))
  File "/Users/jeremy/Seafile/misc_projects/amaranth_playing/venv/lib/python3.9/site-packages/amaranth/sim/_pyrtl.py", line 463, in __call__
    processes.update(self(subfragment))
  File "/Users/jeremy/Seafile/misc_projects/amaranth_playing/venv/lib/python3.9/site-packages/amaranth/sim/_pyrtl.py", line 455, in __call__
    exec(compile(code, filename, "exec"), exec_locals)
RecursionError: maximum recursion depth exceeded during compilation
```

closed 2024-02-09 by whitequark