amaranth-lang/amaranth #360: nMigen Linter

awygle, 2020-04-20:
It would be great to have a linter which reports on things like "you are assigning signals of different widths", "you assign this comb signal in one branch of this switch but not another", or "your FSM has multiple exit transitions with the same condition".

whitequark, 2020-04-20:
My current plan for addressing this need is providing some example linters, and if the idea turns out to be popular, we can figure out some nice way to integrate them with core nMigen.

Maykeye, 2022-07-30:
Regarding width, fortunately Signal is not final (`@final` is [commented out](https://github.com/amaranth-lang/amaranth/blob/db49294cf722e9463666bd06f81a86f930b8707c/amaranth/hdl/ast.py#L896) ). So for now at least it's possible to workaround by wrapping signals in python code directly like

    class SafeSignal(Signal): 
     def __eq__(self, other):
        self.__check_shapes(other)
        return super().__eq__(other)

     def __check_shapes(self, other : Value):
        if hasattr(other, "shape"):
            if other.shape().width != self.shape().width:
                raise ValueError(f"{self}@{self.src_loc} and {other}@{other.src_loc} have different shapes: {self.shape()} vs {other.shape()}")
   
    ...
    a : Signal = SafeSignal(8)

It will not catch all errors (`Signal(8)==1024`) and requires overriding lots of methods to be effective, but at least it can save from some headache

whitequark, 2022-07-30:
> Regarding width, fortunately Signal is not final (`@final` is [commented out](https://github.com/amaranth-lang/amaranth/blob/db49294cf722e9463666bd06f81a86f930b8707c/amaranth/hdl/ast.py#L896) ).

This will stop being the case once `amaranth.compat` is deprecated for good. We'll need to come up with a different design by that point, probably.

whitequark, 2024-02-09:
Adding a linter would need an RFC. In general, I favor good language design and built-in (always-on) diagnostics over linting, but it seems inevitable that someone will want a linter, too. This isn't something that's currently on the roadmap, so I'll go ahead and close this issue.

If you strongly feel that there should be a linter, please comment on this issue and/or [write an RFC](https://amaranth-lang.org/rfcs/).

closed 2024-02-09 by whitequark