amaranth-lang/amaranth #374: Bad error message when using Signal in add_clock_constraint not actually used in the design

rroohhh, 2020-04-30:
This snippet:

```python
from nmigen import *
from nmigen_boards.zturn_lite_z010 import *

class Test(Elaboratable):
    def elaborate(self, plat):
        m = Module()

        clk = Signal()
        plat.add_clock_constraint(clk, 100_000_000)

        return m

plat = ZTurnLiteZ010Platform()

t = Test()

plat.build(t)
```

causes the following error:
```
Traceback (most recent call last):
  File "clock_constraint_vivado.py", line 18, in <module>
    plat.build(t)
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/nmigen/build/plat.py", line 90, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/nmigen/build/plat.py", line 163, in prepare
    return self.toolchain_prepare(fragment, name, **kwargs)
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/nmigen/build/plat.py", line 411, in toolchain_prepare
    render(content_tpl, origin=content_tpl))
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/nmigen/build/plat.py", line 405, in render
    "autogenerated": autogenerated,
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "<template>", line 12, in top-level template code
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/nmigen/build/plat.py", line 369, in hierarchy
    return separator.join(name_map[signal][1:])
  File "/home/robin/.guix-profile/lib/python3.7/site-packages/nmigen/hdl/ast.py", line 1440, in __getitem__
    return self._storage[key]
KeyError: <nmigen.hdl.ast.SignalKey (sig clk)>
```

This is probably not likely to happen in a real design, but it is still not a really useful error message.

whitequark, 2020-04-30:
This kind of thing is actually fairly likely in a generic design with a bunch of it disabled at elaboration time, so it's very important to fix it.

rroohhh, 2020-04-30:
Ah makes perfect sense.

whitequark, 2020-05-02:
Hm, this is somewhat annoying to fix because `iter_clock_constraints` doesn't know if the net ended up as a part of the design or not.

whitequark, 2020-05-02:
Architecturally, I'm not sure what the best way to fix this is. I'm going to avoid thinking about this until I get around to fixing this issue, or if someone has a burning desire to get it to work they can ask here, too.

commit 3420b683 (2020-05-20): build.plat: skip clock constraints on unused signals.

It's not very nice to add more internal mutable state to Platform
related classes, but our whole approach for Platform is inherently
stateful, and other solutions (like changing every individual vendor
platform to check for unused signals) are even worse.

Fixes #374.
closed 2020-05-20 by whitequark