amaranth-lang/amaranth #375: Net names are not escaped properly for vivado tcl

rroohhh, 2020-04-30:
It seems like net names are not properly escaped for vivado tcl scripts, atleast `$` is causing problems.

This example:
```python
from nmigen import *
from nmigen_boards.zturn_lite_z010 import *

class Test(Elaboratable):
    def elaborate(self, plat):
        m = Module()

        dummy1 = Signal()

        clk = Signal(name="clk$1")
        m.d.comb += dummy1.eq(clk)
        plat.add_clock_constraint(clk, 100_000_000)

        return m

plat = ZTurnLiteZ010Platform()

t = Test()

plat.build(t)
```

causes vivado to emit the following message:
```
Parsing XDC File [/data/projects/nmigen_bugs/build/top.xdc]
CRITICAL WARNING: [Common 17-1548] Command failed: can't read "1": no such variable
 [/data/projects/nmigen_bugs/build/top.xdc:2]
Finished Parsing XDC File [/data/projects/nmigen_bugs/build/top.xdc]
```

`dummy1` is merely used to work around / avoid #374.

Vivado fails later due to the design being empty, but I don't think that matters for this bug.

Full build folder: [build.zip](https://github.com/nmigen/nmigen/files/4559517/build.zip)

rroohhh, 2020-04-30:
Playing around a bit with vivado makes me come to the conclusion, that atleast `'"{};$#[]\` are special characters and need to be escaped using `\`.

Furthermore there seem to be different rules for clock names, alluded to by this error message:

```
CRITICAL WARNING: [Vivado 12-2270] Clock names may not contain tcl special characters: '"{};$# - Skipping 'clk$1' [/data/projects/nmigen_bugs/build/top.xdc:2]
```

Which gets emitted when changing the constraint file to this:
```
create_clock -name clk\$1 -period 10.0 [get_nets \\clk\$1\ ]
```

whitequark, 2020-04-30:
I -think- we can get away with using `{` `}` around net names and escaping *only* `{` and `}`. (Since we can't guarantee that we'll have a balanced number of those in a signal name, naturally.)

rroohhh, 2020-04-30:
Atleast `\` still needs escaping in the case of using `{` `}` around net names, atleast if it is the last character of the name.

whitequark, 2020-04-30:
Yup that's true.

Regarding escaping itself, it'll have to be fully manual. You'd *think* Jinja2 will make it possible to customize the escaping engine, but nope! Hardcoded to HTML/XML.

rroohhh, 2020-04-30:
Ouch, that is sad.

commit 6cee2804 (2020-05-02): plat, vendor: systematically escape net and file names in Tcl.

Before this commit, there was only occasional quoting of some names
used in any Tcl files. (I'm not sure what I was thinking.)

After this commit, any substs that may include Tcl special characters
are escaped. This does not include build names (which are explicitly
restricted to ASCII to avoid this problem), or attribute names (which
are chosen from a predefined set). Ideally we'd use a more principled
approach but Jinja2 does not support custom escaping mechanisms.

Note that Vivado restricts clock names to a more restrictive se [... cut]
closed 2020-05-02 by whitequark