amaranth-lang/amaranth #384: Catch matching on an enum of the wrong type

ZirconiumX, 2020-05-11:
If you have `x = Signal(decoder=Foo)` and try to do `x.match(Bar.QUUX)` or `m.Switch(x)`/`m.Case(Bar.QUUX)`, there should be an exception (`TypeError`?) thrown to catch any potential bugs where an enum of the wrong type is being used.

whitequark, 2023-01-31:
We can do this on a best-effort basis with a lint (warning by default diagnostic), but Amaranth is an untyped language (in that all values belong to the same type and are mutually interoperable regardless of how they're constructed) and it is not feasible to add a type system to it even if it was desired.

whitequark, 2023-01-31:
This doesn't change the language and so doesn't need an RFC, someone can just implement it. @Ravenslofty?