amaranth-lang/amaranth #386: Inverted DiffPairs in connectors

anuejn, 2020-05-14:
I have a board that has some inverted differential pairs broken out to a connector. The connector is used on several boards and there are daughterboards that can be plugged into those connectors.

Right now I am modeling this using the nmigen platform dsl in the following way: My board defines connectors. The add-on boards are functions which take the platform as an argument and then add the resources they hold to the connectors. For example:
```python
def hdmi_plugin_connect(platform, plugin_number):
    platform.add_resources([
        Resource("hdmi", plungin_number,
             # high speed serial lanes
             Subsignal("clock", DiffPairs("lvds3_p", "lvds3_n", dir='o', conn=("plugin", plugin_number)), Attrs(IOSTANDARD="LVDS_25")),
             Subsignal("data", DiffPairs("lvds2_p lvds1_p lvds0_p", "lvds2_n lvds1_n lvds0_n", dir='o', conn=("plugin", plugin_number)), Attrs(IOSTANDARD="LVDS_25"))
        )
    ])
```
In my platform i do:
```python
self.add_connectors([
    Connector(
        "plugin", "north",
        {
            "lvds0_p": 21, "lvds0_n": 23,
            "lvds1_p": 3, "lvds1_n": 5,
            "lvds2_p": 9, "lvds2_n": 11,
            "lvds3_p": 13, "lvds3_n": 15,
        },
        conn=("expansion", 0),
    )
])
```

Now one of the Lanes on the mainboard (not the daughterboard) is inverted (lvds3_p is connected to the negative side of the FPGAs differential buffer and vice versa to ease routing on the PCB). This raises the requirement, that I somehow need to mark pins as inverted diff pairs right in the platform and to mark single pairs of a DiffPairs Object as inverted.

How would you model this using the existing platform dsl or what would be the preferred way to extend the dsl?

whitequark, 2020-05-14:
No idea. I never considered this case and the existing DSL isn't built to support it.

peteut, 2020-05-17:
@anuejn Use the vendor macro, so for Xilinx Series7 this would be `IBUFGDS_DIFF_OUT`, then connect use `OB` instead of the `O` output as documented in [1].

[image]

[1] https://www.xilinx.com/support/documentation/user_guides/ug471_7Series_SelectIO.pdf

whitequark, 2020-05-17:
The question is about the platform DSL. Of course you can always bypass the DSL, but it would be nice if we didn't have to.

whitequark, 2020-05-17:
I suspect the best way to handle this (hopefully very rare!) corner case is to allow passing a tuple to `invert=` rather than a single bool, but I haven't thought very deeply about this.

anuejn, 2020-10-20:
Yup that would be quite nice :)

anuejn, 2020-10-20:
Should I formulate a separate Issue as an RFC or is this enough space for discussion on that specific topic?

whitequark, 2020-10-20:
Ideally an RFC would be a separate issue, though it's not that important here. (It mostly matters when there's a lot of discussion to follow, since a separate issue allows summarizing all that without forcing people to skip unnecessary context.)