amaranth-lang/amaranth #388: Integer parameters over 32 bits

dlharmon, 2020-05-17:
6e29fbcc61ed8e8ebdad8d64bda1ba52343fa296 breaks integer parameters greater than `2**32 -1`

With Yosys HEAD, a large integer parameter results in the error `nmigen.back.verilog.YosysError: ERROR: Parser error in line 247: syntax error`, with older Yosys, the parameter is silently clipped to `2**32-1`.

I'm not sure what the best way to handle this is. It could be as simple as converting large integer parameters to ast.Const parameters. Even an assert to catch this early may be an option.

whitequark, 2020-05-17:
> I'm not sure what the best way to handle this is.

Plain integers used as parameters in RTLIL are equivalent to 32-bit constants, so we can convert them on nMigen's side without loss of generality.

whitequark, 2020-05-18:
> breaks integer parameters greater than `2**32 -1`

> with older Yosys, the parameter is silently clipped to `2**32-1`.

I think it's even worse than this. Consider this RTLIL:

```
module \foo
	attribute \init 2147483647
	wire width 64 \2**31-1
	#attribute \init 2147483648
	#wire width 64 \2**31
	attribute \init -2147483647
	wire width 64 \-2**31+1
	attribute \init -2147483648
	wire width 64 \-2**31
end
```
If you convert it to Verilog, Yosys outputs:
```v
module foo();
  (* init = 32'd2147483648 *)
  wire [63:0] \-2**31 ;
  (* init = 32'd2147483649 *)
  wire [63:0] \-2**31+1 ;
  (* init = 32'd2147483647 *)
  wire [63:0] \2**31-1 ;
endmodule
```

So it's discarding the sign bit!

whitequark, 2020-05-18:
Looking closer, it seems that negative numbers in RTLIL don't roundtrip:

```
module \foo
  attribute \init 32'10000000000000000000000000000000
  wire width 64 \-2**31
  attribute \init 32'10000000000000000000000000000001
  wire width 64 \-2**31+1
  attribute \init 2147483647
  wire width 64 \2**31-1
end
```

whitequark, 2020-05-18:
Proposed upstream fix: https://github.com/YosysHQ/yosys/pull/2066

commit fbf9e1f3 (2020-05-19): back.rtlil: handle signed and large Instance parameters correctly.

Fixes #388.
closed 2020-05-19 by whitequark