amaranth-lang/amaranth #395: add_clock_constraint not working

andresdemski, 2020-06-02:
Hi! 
I'm trying to add ZynqMP PS to an nmigen design I don't need to  create a block design again.

I wrapped PS instantiation with an nMiguen elaboratable and I added a `DONT_TOUCH` constraint in order to avoid deletion during optimization process.

In this case, the clk source comes from a signal so i added  a clock constraint as is show next:

```python
from nmigen import ClockSignal, ResetSignal, ClockDomain
from nmigen.cli import main
from .platform import MyPlatform

class Test(Elaboratable):
    def elaborate(self, platform):
        m = Module()
        m.domains += ClockDomain('sync')
        m.submodules.ps = ps = PsZynqMP()

        cnt = Signal(8)
        m.d.sync += cnt.eq(cnt + 1)
        m.d.comb += ps.get_irq_signal(0).eq(cnt)

        clk = ps.get_clock_signal(0)
        platform.add_clock_constraint(clk, 200e6)
        m.d.comb += ResetSignal().eq(~ps.get_reset_signal(0))
        m.d.comb += ClockSignal().eq(clk)

        return m

core = Test()
plat = MyPlatform()
constraints = '''
set_property DONT_TOUCH true [get_cells  "ps/PS8_i"]
'''
plat.add_file('zynq_ultra.v', open('zynq_ultra.v').read())
plat.build(core, add_constraints=constraints)
```
the xdc file is generated correctly:

```
# Automatically generated by nMigen 0.2. Do not edit.
create_clock -name pl_clk0 -period 5.0 [get_nets pl_clk0]
set_property DONT_TOUCH true [get_cells  "ps/PS8_i"]% 
```

But vivado is not recognizing the constraint:

```
CRITICAL WARNING: [Vivado 12-4739] create_clock:No valid object(s) found for '-objects [get_nets pl_clk0]'.
```

I added a constrain manually and it works as it is expected:
```
constraints = '''
set_property DONT_TOUCH true [get_cells  "ps/PS8_i"]
create_clock -name clk_pl_0 -period "10" [get_pins "ps/PS8_i/PLCLK[0]"]
 '''
plat.add_file('zynq_ultra.v', open('zynq_ultra.v').read())
plat.build(core, add_constraints=constraints)
```

In vivado, the command `get_nets pl_clk0` doesn't work. `pl_clk0` is in the verilog file so it looks like it was simplified:

[image]

whitequark, 2020-06-02:
Does it work if you replace `keep` with `dont_touch` here?
https://github.com/nmigen/nmigen/blob/9c80c32c306c4cd9483821e3503141111ed7ffde/nmigen/vendor/xilinx_ultrascale.py#L180-L182

andresdemski, 2020-06-03:
I was using nMigen from pypi, with last master it works perfectly. Sorry.

closed 2020-06-03 by andresdemski