amaranth-lang/amaranth #396: Add attributes/contraints to a module

andresdemski, 2020-06-03:
Hi,

In order to add PS8 instance (for ZynqMP devices) to nMigen project, i need to add the following constraint:
```
set_property DONT_TOUCH true [get_cells  "ps/PS8_i"]
```

The only way i found was:

```python
core = Test()
plat = MyPlatform()
constraints = '''
set_property DONT_TOUCH true [get_cells  "ps/PS8_i"]
'''
plat.build(core, add_constraints=constraints)
```

Is there any other way to do this? How can i add a constraint/attribute to a module in the `elaborate` method?

whitequark, 2020-06-03:
Do you actually want to add the attribute to an *instance*? In that case try `Instance("PS8", a_DONT_TOUCH="true", ...)`.

andresdemski, 2020-06-03:
Works like a charm. Thanks you!

closed 2020-06-03 by andresdemski