amaranth-lang/amaranth #397: Wrong number of pins reported when the direction is "io"

jeanthom, 2020-06-03:
Hi,

I just came across a bug that causes nMigen to report the wrong number of pins when the direction of the pins is set to bidirectional.

Test code:
```python
from nmigen import *
from nmigen_boards.arty_a7 import ArtyA7Platform

platform = ArtyA7Platform()
ram = platform.request("ddr3", 0)

print("Number of DQ signals:", len(ram.dq)) # DQ is "io"
print("Number of A signals:", len(ram.a)) # A is output only
```

Result:
```
Number of DQ signals: 33
Number of A signals: 14
```

What I was expecting:
```
Number of DQ signals: 16
Number of A signals: 14
```

whitequark, 2020-06-03:
You should never use attributes of the object returned by `platform.request` directly. Instead, use `ram.x.i` for inputs or inouts, `ram.x.o` for outputs or inouts, and `ram.x.oe` for inouts.

jeanthom, 2020-06-03:
Ok, I will use .i/.o/.oe instead.

closed 2020-06-03 by jeanthom