amaranth-lang/amaranth #403: Records don't work in Assert

BracketMaster, 2020-06-13:
You can't don Assert on records.
The following will fail in elaboration.

```python
from nmigen import *
from nmigen.asserts import Assert

rec = Record([("sig1", 3), ("sig2", 3)])
Assert(rec == 3)
```

whitequark, 2020-11-07:
Works for me:
```
from nmigen import *
from nmigen.asserts import Assert
from nmigen.back import verilog

rec = Record([("sig1", 3), ("sig2", 3)])
m = Module()
m.d.sync += Assert(rec == 3)
print(verilog.convert(m))
```

Can you please confirm?

BracketMaster, 2020-11-07:
Seems fine. I'm honestly not sure what I was trying to 
do in the original issue. Its been four months haha.

rroohhh, 2020-11-07:
Due to #533, while it works it probably does not produce what one expects:

```python
>>> from nmigen import *
>>> from nmigen.asserts import Assert
>>>
>>> rec = Record([("sig1", 3), ("sig2", 3)])
>>> Assert(rec == 3)
(assert (const 1'd0))
```

whitequark, 2020-11-07:
@rroohhh Thanks for noticing!

rroohhh, 2020-11-09:
Should be fixed by #541 now:
```python
>>> from nmigen import *
>>> from nmigen.asserts import Assert
>>> rec = Record([("sig1", 3), ("sig2", 3)])
>>> Assert(rec == 3)
(assert (== (cat (sig rec__sig1) (sig rec__sig2)) (const 2'd3)))
>>>
```
and
```python
from nmigen import *
from nmigen.asserts import Assert
from nmigen.back import verilog

rec = Record([("sig1", 3), ("sig2", 3)])
m = Module()
m.d.sync += Assert(rec == 3)
print(verilog.convert(m))
```

```verilog
/* Generated by Yosys 0.9+3477 (git sha1 UNKNOWN, g++ 7.5.0 -fPIC -Os) */

(* \nmigen.hierarchy  = "top" *)
(* top =  1  *)
(* generator = "nMigen" *)
module top(rec__sig2, rst, clk, rec__sig1);
  reg \initial  = 0;
  (* src = "/data/projects/nmigen/nmigen/hdl/rec.py:255" *)
  wire \$1 ;
  (* src = "<stdin>:1" *)
  reg \$assert$check  = 1'h0;
  (* src = "<stdin>:1" *)
  reg \$assert$check$next ;
  (* src = "<stdin>:1" *)
  reg \$assert$en  = 1'h0;
  (* src = "<stdin>:1" *)
  reg \$assert$en$next ;
  (* src = "/data/projects/nmigen/nmigen/hdl/ir.py:526" *)
  input clk;
  (* src = "<stdin>:1" *)
  input [2:0] rec__sig1;
  (* src = "<stdin>:1" *)
  input [2:0] rec__sig2;
  (* src = "/data/projects/nmigen/nmigen/hdl/ir.py:526" *)
  input rst;
  assign \$1  = { rec__sig2, rec__sig1 } == (* src = "/data/projects/nmigen/nmigen/hdl/rec.py:255" *) 2'h3;
  always @* if (\$assert$en ) assert(\$assert$check );
  always @(posedge clk)
    \$assert$en  <= \$assert$en$next ;
  always @(posedge clk)
    \$assert$check  <= \$assert$check$next ;
  always @* begin
    if (\initial ) begin end
    \$assert$en$next  = \$assert$en ;
    \$assert$check$next  = \$assert$check ;
    \$assert$check$next  = \$1 ;
    \$assert$en$next  = 1'h1;
  end
endmodule
```

closed 2020-11-09 by whitequark