amaranth-lang/amaranth #405: Making an assignment to an input results in unclear assertion error

jeanthom, 2020-06-17:
```python
from nmigen import *
from nmigen_boards.versa_ecp5 import VersaECP5Platform

class Top(Elaboratable):
    def elaborate(self, platform):
        m = Module()
        input = platform.request("eth_clk125", 0) # This is an input
        m.d.comb += input.eq(1)
        return m

if __name__ == "__main__":
    VersaECP5Platform().build(Top())
```

```
Traceback (most recent call last):
  File "bidir.py", line 16, in <module>
    VersaECP5Platform().build(Top())
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/build/plat.py", line 90, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/build/plat.py", line 162, in prepare
    fragment._propagate_ports(ports=self.iter_ports(), all_undef_as_ports=False)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py", line 462, in _propagate_ports
    self._prepare_use_def_graph(parent, level, uses, defs, ios, self)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py", line 436, in _prepare_use_def_graph
    subfrag._prepare_use_def_graph(parent, level, uses, defs, ios, top)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py", line 428, in _prepare_use_def_graph
    add_defs(value._lhs_signals())
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py", line 396, in add_defs
    assert defs[sig] is self
AssertionError
```

Ideally the error message would point to the faulty assignment line.

whitequark, 2020-06-17:
Same underlying problem as #398.

FFY00, 2020-07-14:
I had a look at this, I don't quite understand the context of how this is happening. @whitequark perhaps you could give me some hints?

whitequark, 2020-07-14:
> perhaps you could give me some hints?

I wrote that code a while ago, and it's not very good. It'd probably be easier for me to fix it myself if I have to sit down and remember how it works.

PR #1104 (2024-02-27): ir: kill Fragment.ports
Fixes #191.
Fixes #320.
Fixes #398.
Fixes #405.
Fixes #600.
closed 2024-02-27 by whitequark