amaranth-lang/amaranth #406: Can instantiate a Tristate without input on ECP5

jeanthom, 2020-06-17:
```python
from nmigen import *
from nmigen.build import Resource, Pins, Attrs
from nmigen_boards.versa_ecp5 import VersaECP5Platform
from nmigen.compat.fhdl.specials import Tristate

class Top(Elaboratable):
    def elaborate(self, platform):
        m = Module()

        ddr = platform.request("ddr3", 0, dir={"dq":"-", "dqs":"-"})
        
        output = Signal()
        m.d.comb += output.eq(1)

        dir = Signal()
        m.d.comb += dir.eq(1)

        m.submodules += Tristate(ddr.dq.io[0], output, dir)
        
        return m

if __name__ == "__main__":
    VersaECP5Platform().build(Top())
```

The above example code generates the following error:

```
Traceback (most recent call last):
  File "testpins.py", line 24, in <module>
    VersaECP5Platform().build(Top())
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/build/plat.py", line 90, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/build/plat.py", line 130, in prepare
    fragment = Fragment.get(elaboratable, self)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py", line 39, in get
    obj = obj.elaborate(platform)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/dsl.py", line 541, in elaborate
    fragment.add_subfragment(Fragment.get(submodule, platform), None)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/hdl/ir.py", line 39, in get
    obj = obj.elaborate(platform)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/compat/fhdl/specials.py", line 48, in elaborate
    return platform.get_input_output(pin, self.target, attrs={}, invert=None)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/vendor/lattice_ecp5.py", line 515, in get_input_output
    o_O=i[bit],
TypeError: 'NoneType' object is not subscriptable
```

As a quick workaround you can use Signal() instead of None as the input default value.

whitequark, 2020-06-17:
Why are you using `nmigen.compat` at all here? It shouldn't be used together with new code.

-

jeanthom, 2020-06-17:
I'm using it because I'm trying to migrate oMigen code to nMigen and `nmigen.compat` seems to me the right thing to do (at least until everything has been migrated to nMigen).

whitequark, 2020-06-17:
The right thing in this case would be to figure out exactly what kind of IO you need and then instantiate it. Can you show me the oMigen code?

jeanthom, 2020-06-17:
This: https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/ecp5ddrphy.py

I'm almost certain that I could use the "io" type and control all directions at one for DQ/DQS, but I'd still like to know how to individually set IO directions without defining a subsignal for each pin.

whitequark, 2020-06-17:
> I'd still like to know how to individually set IO directions without defining a subsignal for each pin.

The general answer is: if you actually have to do that to implement an interface, then the board definition (and, in fact, every board definition in nmigen-boards that includes this interface) should be changed to have separate subsignals per pin.

For DDR on ECP5 specifically, I would suggest that you use a `BB` instance, since you're using ECP5-specific cells already, anyway.

On reflection, I also agree that the current implementation of the `Tristate` compatibility shim isn't good enough for some use cases, like yours, and should probably be fixed.

whitequark, 2020-06-17:
(The issue with the `Tristate` implementation affects all platforms, not just ECP5.)

commit 8dd28fec (2020-07-02): compat.fhdl.specials: fix handling of tristate (i=None) pins.

Fixes #406.
closed 2020-07-02 by whitequark