amaranth-lang/amaranth #407: AttributeError: module 'importlib.resources' has no attribute 'files'

colepoirier, 2020-06-20:
It looks like a recent change to nmigen dependencies is causing ModuleNotFoundError's

```
Traceback (most recent call last):
  File "/home/colepoirier/src/nmigen/nmigen/_yosys.py", line 17, in <module>
    importlib_resources.files # py3.9+ stdlib
AttributeError: module 'importlib.resources' has no attribute 'files'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/colepoirier/src/nmigen/nmigen/_yosys.py", line 19, in <module>
    import importlib_resources # py3.8- shim
ModuleNotFoundError: No module named 'importlib_resources'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "decoder/pseudo/pywriter.py", line 8, in <module>
    from soc.decoder.power_pseudo import convert_to_python
  File "/home/colepoirier/src/soc/src/soc/decoder/power_pseudo.py", line 18, in <module>
    from soc.decoder.power_decoder import create_pdecode
  File "/home/colepoirier/src/soc/src/soc/decoder/power_decoder.py", line 84, in <module>
    from nmigen.cli import rtlil
  File "/home/colepoirier/src/nmigen/nmigen/cli.py", line 4, in <module>
    from .back import rtlil, cxxrtl, verilog, pysim
  File "/home/colepoirier/src/nmigen/nmigen/back/cxxrtl.py", line 1, in <module>
    from .._yosys import *
  File "/home/colepoirier/src/nmigen/nmigen/_yosys.py", line 21, in <module>
    import importlib_resources # py3.6- shim
ModuleNotFoundError: No module named 'importlib_resources'
```

whitequark, 2020-06-20:
How are you installing nMigen?

colepoirier, 2020-06-21:
`python3 setup.py develop`

It appeared in the past that it was only necessary to run this command once, and new commits pulled in after this would work fine. Now, it appears that one must run `python3 setup.py develop` every time a new commit is pulled in from nmigen/nmigen source repo.

whitequark, 2020-06-21:
> Now, it appears that one must run `python3 setup.py develop` every time a new commit is pulled in from nmigen/nmigen source repo.

Are you sure? I think what happened in your case is that when you update, the files are automatically updated as well, but the *dependencies* are not; to get those updated you have to rerun `setup.py`.

colepoirier, 2020-06-21:
That sounds accurate to what happens. The effect however is the same, `python3 setup.py develop` must be run every time new nmigen code is pulled in case there have been dependency changes. This is fine now that I know this is the assumption I must operate under. Is there a note to this effect on the README? If not, I think it is probably a necessary addition. Thanks for your help in clearing this up.

whitequark, 2020-06-21:
There isn't; I'm going to describe this in the upcoming documentation (see the `doc` branch) which already goes into more detail with regards to Python package management.

colepoirier, 2020-06-21:
Ah thanks very much. Sorry I was looked in the docs older of the master branch, now having looked at the same under the 'doc' branch things are much clearer. I really appreciate you taking the time and effort to do this!

whitequark, 2020-06-21:
> Sorry I was looked in the docs older of the master branch, now having looked at the same under the 'doc' branch things are much clearer.

Yeah the new docs aren't really finished or properly public yet, it's understandable you'll get confused.

whitequark, 2020-06-30:
This is now documented at https://nmigen.info/nmigen/latest/install.html#development-snapshot.

closed 2020-06-30 by whitequark