amaranth-lang/amaranth #412: LICENSE.txt and copyright

Fatsie, 2020-06-24:
Currently the LICENSE.txt file contains only copyright for M-Labs:
```
Copyright (C) 2011-2019 M-Labs Limited
```
I would like that it is made clear that the copyright of each commit is held by it's author or something like that.
I think this issue if valid also for the other nmigen-* repos.

whitequark, 2020-06-24:
Yeah, we should fix that.

mithro, 2020-06-24:
It would also be good if GitHub correctly identified the license in the repo.

whitequark, 2020-07-01:
> I would like that it is made clear that the copyright of each commit is held by it's author or something like that.
> I think this issue is valid also for the other nmigen-* repos.

Is there some boilerplate I can use?

> It would also be good if GitHub correctly identified the license in the repo.

Looks like the GitHub `licensee` gem cannot identify the 2-clause BSD license we're using. Any advice?

alanvgreen, 2020-07-01:
When I remove the final paragraph of LICENSE.txt, `licensee` recognizes the remaining text as the 2-clause BSD license.

IANAL, but...

It seems the final paragraph is not part of the license. It could be moved to a section of the README.md or some other file.

whitequark, 2020-07-01:
Hm, I see. Should we take a page from Rust's book? See their [LICENSE-MIT](https://github.com/rust-lang/rust/blob/master/LICENSE-MIT) and [COPYRIGHT](https://github.com/rust-lang/rust/blob/master/COPYRIGHT) files.

alanvgreen, 2020-07-01:
I think that would work for github, and also give room to explain/document anything you wanted about copyrights and licensing.

I was reading the final paragraph again: "All submissions are strongly encouraged to use the two-clause BSD license reproduced above." If it were me, I would insist on all contributions to this repo being 2-clause BSD licensed, because I don't understand the implications of mixing 2-clause BSD and other licenses. It sounds complicated and boring. However, this is just my opinion - IANAL, am definitely operating outside my area of expertise, and it's not my IP.

Happy to put together a patch if you give me a general direction.

whitequark, 2020-07-01:
> I would insist on all contributions to this repo being 2-clause BSD licensed, because I don't understand the implications of mixing 2-clause BSD and other licenses

Yes, we should require that. That clause was cargo-culted from Migen and really shouldn't have ever been there. There is no code that isn't under 2-clause BSD anywhere in nMigen, anyway.

Please do prepare a PR!

mithro, 2020-07-01:
@whitequark - The advice I have been given by lawyers are;
 - You should not add anything beyond the OSI approved text for a license to the LICENSE file (hence why licensee are so strict).
 - Information like what you are adding should be in something like the CONTRIBUTING or README files.
 - If you have an "Authors" or "Contributing" like file, you should add text like the following;
```
# This is the list of XXXXs significant contributors.
#
# This does not necessarily list everyone who has contributed code,
# especially since many employees of one corporation may be contributing.
# To see the full list of contributors, see the revision history in
# source control.
```
 - If you want fine grain information, you should use SPDX identifiers and full license info at the header of the code file.

whitequark, 2020-07-01:
Thanks. This aligns with what Rust is doing. Let's copy them.

whitequark, 2020-07-08:
Fixed in 3a4576e033bbed88541415e6a747c328cdd1f420.

closed 2020-07-08 by whitequark