amaranth-lang/amaranth #413: Per-pin direction control for platform defined IOs

jeanthom, 2020-06-25:
In [LiteDRAM](https://github.com/enjoy-digital/litedram), each DQ pin has its OE signal driven by its own TSHX2DQSA instance. There are currently two ways of doing this in nMigen:
 * Declare a bidirectional subsignal for each IO: nMigen takes care of the bidirectional IO initialization, but you end up with dozens of independent subsignals
 * Using `dir={"mysubsignal":"-"}` when `platform.request` is called: you have to declare the bidirectional IO instances yourself

We're not sure that we actually have to declare TSHX2DQSA for each pin, but that might be useful for other protocols (which ones?).

whitequark, 2020-06-25:
> * Using `dir={"mysubsignal":"-"}` when `platform.request` is called: you have to declare the bidirectional IO instances yourself

Can you actually do this? I recall that you ran into some trouble where the `TSHX2DQSA` instances had to be in the toplevel module, which isn't easily expressible in nMigen. Is that right?

jeanthom, 2020-06-25:
Yep, it turned out to be an issue in my code. It does synthesize correctly in Yosys/Nextpnr and Diamond.

whitequark, 2020-06-25:
Then I would say that using `dir="-"` is the proper solution here. It works and it matches LiteDRAM code; is there any reason you want to avoid it?

jeanthom, 2020-06-25:
What's bothering me is that it is generally the responsibility of the end user to write the `platform.request` code. Ideally you'd want the end user to deal with the least amount of platform dependant code so that their code is portable across FPGA boards or even FPGA architectures.

whitequark, 2020-06-25:
Of course. But the entire reason you are having this problem is that you are using the platform specific primitive `TSHX2DQSA`, right? So it makes no sense to try and abstract over this stuff if in the end you're going to break the abstraction anyway.

whitequark, 2020-07-31:
Closing in favor of #458.

closed 2020-07-31 by whitequark