amaranth-lang/amaranth #417: Can't emulate fully combinatorial design

jeanthom, 2020-07-01:
Hi! I noticed that the following code can't be simulated with pysim:

```python
from nmigen import *
from nmigen.cli import main

class Top(Elaboratable):
	def __init__(self):
		self.a = Signal()

	def elaborate(self, platform):
		m = Module()

		with m.FSM():
			with m.State("aaaaaaaa"):
				m.d.comb += self.a.eq(1)

		return m

if __name__ == "__main__":
	top = Top()
	main(top, ports=[top.a])
```

I get the following error message:

```
[jeanthomas@computer test]$ python test.py simulate -c 10
Traceback (most recent call last):
  File "test.py", line 20, in <module>
    main(top, ports=[top.a, top.b])
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/cli.py", line 78, in main
    main_runner(parser, parser.parse_args(), *args, **kwargs)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/cli.py", line 71, in main_runner
    sim.add_clock(args.sync_period)
  File "/home/jeanthomas/.local/lib/python3.8/site-packages/nmigen/back/pysim.py", line 1005, in add_clock
    raise ValueError("Domain {!r} is not present in simulation"
ValueError: Domain 'sync' is not present in simulation
```

As I workaround I added a sync statement to the design and it worked.

whitequark, 2020-07-01:
Well, what's the point of having a clock in a fully combinatorial design? This is one of the reasons why the CLI `simulate` command was a terrible idea that I regret...

whitequark, 2020-07-01:
I'm closing this with the understanding that the current "toplevel" (`nmigen.cli`) is very primitive and is not suitable for all but the simplest, most ordinary tasks. We'll improve it, but not right now.

closed 2020-07-01 by whitequark