amaranth-lang/amaranth #419: nMigen-Yosys isn't detected by require_tool

jeanthom, 2020-07-02:
Traceback:

```
Traceback (most recent call last):
  File "simsoc.py", line 216, in <module>
    platform.build(soc, build_dir="build_simsoc")
  File "/home/build/.local/lib/python3.7/site-packages/nmigen/build/plat.py", line 88, in build
    require_tool(tool)
  File "/home/build/.local/lib/python3.7/site-packages/nmigen/_toolchain.py", line 36, in require_tool
    format(name, env_var))
nmigen._toolchain.ToolNotFound: Could not find required tool yosys in PATH. Place it directly in PATH or specify path explicitly via the YOSYS environment variable
```

This happens when Yosys is requested in a platform class.

whitequark, 2020-07-02:
That's intended behavior--nmigen-yosys cannot be used for synthesis (in the platform). It can only be used for Verilog and CXX conversion (in the backends).

whitequark, 2020-07-02:
Did you expect to be able to use nmigen-yosys for synthesis based on documentation or for some other reason? The [README](https://github.com/nmigen/nmigen-yosys) and the PyPI package description should mention that, maybe we should mention it elsewhere?

whitequark, 2020-07-02:
Oh yeah and if you'd like to use a full build Yosys that *can* be used for synthesis, check out [YoWASP](https://yowasp.org/), which also includes PNR tools.

jeanthom, 2020-07-02:
I thought it would work for platforms that only rely on verilog code generation in Yosys (like ECP5 with Diamond) but I wasn't considering the platforms that use yosys for their synthesis. That means if we want to be able to use nmigen-yosys in platforms we'd have to specify if we need a Yosys version that has the features we need, which will add more code than reasonable => I'm closing this issue.

Regarding YoWASP, how can you use it with nMigen since the executables are prefixed?

whitequark, 2020-07-02:
> I thought it would work for platforms that only rely on verilog code generation in Yosys (like ECP5 with Diamond) but I wasn't considering the platforms that use yosys for their synthesis.

It is actually supposed to work for that--are you in fact using Diamond? If you are, then this is a bug and we should reopen it.

> Regarding YoWASP, how can you use it with nMigen since the executables are prefixed?

`YOSYS=yowasp-yosys NEXTPNR_ECP5=yowasp-nextpnr-ecp5 ECPPACK=yowasp-ecppack python3 ...`

jeanthom, 2020-07-02:
Nope but I'm using a custom platform based off the ECP5 platform definition (I need a Verilog file with all the I/O and specials cells instantiated that I can feed in Icarus Verilog with Diamond's Verilog models to emulate everything on my FPGA board). Diamond users should theoretically face the same issues as I do because the Yosys executable they use to generate verilog comes from the "require_tool" function (https://github.com/nmigen/nmigen/blob/master/nmigen/vendor/lattice_ecp5.py#L178-L182).

I'm okay with using env variables for setting which Yosys executable to use, I think it's a good approach.

whitequark, 2020-07-02:
> because the Yosys executable they use to generate verilog comes from the "require_tool" function (https://github.com/nmigen/nmigen/blob/master/nmigen/vendor/lattice_ecp5.py#L178-L182).

Ah right, so I think the problem here is that `_diamond_required_tools` is outdated.

commit 6bfff25e (2020-07-02): vendor: `yosys` is not a required tool for proprietary toolchains.

Since commit b9799b4c, the discovery mechanism for the Yosys required
to produce Verilog is different from the usual require_tool(); namely
it is possible to produce Verilog without a `yosys` binary on PATH.

Fixes #419.
closed 2020-07-02 by whitequark