amaranth-lang/amaranth #421: Adding Shapes gives counterintuitive results

awygle, 2020-07-04:
`Shape`s implement `__add__`, but adding them is currently not useful:

```python
>>> from nmigen import *
>>> x = Signal(16)
>>> y = Signal(32)
>>> s = x.shape() + y.shape()
>>> s
(16, False, 32, False)
```

The expected result here (I argue) would be `(48, False)`, but (probably because `Shape` is a named tuple under the hood) we get the not-very-useful result shown above.

whitequark, 2020-07-04:
So there are two unrelated issues here. First, that currently you can do `shape + shape` and the result is never desirable. We should fix that by not using `namedtuple` while still allowing to destructure shapes using `width, signed = shape`. (Is that possible? I think we can override `__iter__` to do it...)

Second, the idea of overloading the `+` operation on shapes to do something not really related to addition. It doesn't seem like a good idea to me, but we can discuss it in a different issue.

awygle, 2020-07-04:
I agree it's a bad idea. We should stop implementing extraneous operators for `Shape`, and that's all.

awygle, 2020-07-06:
Should we preserve comparing `Shape`s to tuples of `(int, bool)`? This currently works (and is used in the test suite) but feels like a bit of a footgun.

whitequark, 2020-07-06:
> This currently works (and is used in the test suite)

Where?

whitequark, 2020-07-06:
Oh sorry, you mean comparing *for equality*, not for ordering. Yeah let's preserve that, we can always deprecate it later if it turns out to cause issues.

commit 659b0e81 (2020-07-07): hdl.ast: don't inherit Shape from NamedTuple.

Fixes #421.
closed 2020-07-07 by whitequark