amaranth-lang/amaranth #427: Add support for Assert in pysim

jeanthom, 2020-07-09:
Adding `Assert` support in pysim would be helpful for migrating LiteX code that relies on Display statements. While not equivalent to the Display statement, it would give enough information to figure out what's happening inside an elaboratable without exposing every single internal signal.

whitequark, 2020-07-31:
(Tagging upstream because the Yosys cxxrtl backend needs to be taught about asserts, too.)

jeanthom, 2020-08-26:
This issue was discussed on August 16th as part of the weekly IRC meeting: https://freenode.irclog.whitequark.org/nmigen/2020-08-17#1597691839-1597695703;

In short:
 * We should use AssertionError (or an exception inherited from AssertionError if needed)
 * Exceptions should not be thrown deep inside generated pysim code. We should instead generate events that will be handled so that we don't break the VCD generation
 * We should append to tracebacks to make errors more legible (polyfill for Python 3.6: https://github.com/pallets/jinja/blob/c074fd5ecb7af70b2f864c9ef545fffd38dc203b/src/jinja2/debug.py)
 * Assume should do the same things as Assert

whitequark, 2024-02-01:
Support for capturing assertions in CXXRTL is about to land upstream as part of https://github.com/YosysHQ/yosys/pull/4128.

Amaranth will require modification to emit `$check` cell from `Assert` statements once that PR lands. The format string within `$check` cell is the same as one within `$print` cell, which will be the basis of the implementation of #432.

whitequark, 2024-03-04:
Support for this feature has been accepted via [RFC 50](https://amaranth-lang.org/rfcs/0050-print.html) and implementation is tracked in issue #1186.

closed 2024-03-04 by whitequark