amaranth-lang/amaranth #429: values in vcd are zero despite simulation's nonzero values

programmerjake, 2020-07-10:
Still haven't narrowed this one down to a reproducer that's all that useful:
http://lists.libre-riscv.org/pipermail/libre-riscv-dev/2020-July/008506.html

used revisions:
nmigen : 30e2f91176edcd1c8766c2cb11f413b9c77936b9
ieee754fpu : 610b4a381e70f45f5684cc281398ce77fb5441fa
nmutil : 3853df675a1e1db24950945f66b076266a7da409
soc : caceb716e9417ed8731ef08b7b260a4c077186b2

ieee754fpu, nmutil, and soc can be cloned using:

```bash
for i in ieee754fpu nmutil soc; do
    git clone --recursive git://git.libre-riscv.org/$i.git
done
```

activate a python3 venv:
```bash
<path/to/venv>/bin/activate
```

in each of nmigen, ieee754fpu, and nmutil, run:
```bash
python setup.py develop
```

in soc run:
```bash
make install
python src/soc/fu/div/test/test_pipe_caller.py # will fail the tests
```

the generated vcd file is `div_simulator.vcd` in soc's top level folder.

programmerjake, 2020-07-10:
values that should be nonzero because `print` calls produce the right values:
path in vcd: `top.alu.pipe_start.setup_stage.ra`

whitequark, 2020-07-10:
Can you bisect nmigen to find the culprit?

programmerjake, 2020-07-11:
in case anyone's interested, here's the script I'm using to bisect it:
http://lists.libre-riscv.org/pipermail/libre-riscv-dev/2020-July/008532.html

programmerjake, 2020-07-11:
Well, I got only partially useful results:

git bisect's output:

There are only 'skip'ped commits left to test.
The first bad commit could be any of:
ba1b4c20c83e1c0a1a5ac9387f9aa105f46e14c1
14845af235eb815e8529796833d6a394c54205ab
d5d3bc4b1aba8dc0204268561c342b41c42c6617
102565f6de3ae0f6918b7ed902d29b471388ba9a
6bfff25e7614399a3e660f72d964e100172e9416
369bc3e30721eb48e4425c8951cf9d99a4ab7e18
75a466537ee821dd5e082c69100142d6c7a6274b
7fa4b348a954d2a94925f733fc5f3825cfa33b84
8dd28fecc6348474d61d57fda5eedfbb78cdb7e9
6c0530cf4e0c5688ec5b6e408aa1e04abe0f1815
fd5ee548b6e333c92877440524c116e09bdd7c73
416b2531aed20ce7e661f2a211c2fb76311fe83b
76b3ef4b13b048d4d089fb72da922c1bee49562b
76c7e70582b5540565482b474242b0ee17a769c9
f24f1b02e871d8c7af9ead26c615645ccfcb0016
126f0be731ab008657ca1c4fc05ce7cc6e355169
f1153352c0e0d6c1e5ff345e5fbab5653bb8f24c
1fbd7f1d08b964131b8bdb772817b7765938274e
f5670a9b719a2c240949df853b5425ceea8244d0
175c8a596ef1adaa74961a47e48a0ed441c05809
43da4e3f0a356c356c7f2ddcf8ad7963b9daf5a1
981e67408111fbfcf9bd0df1f41feeb03212ce12
f50303ca7c46bca66f454b4de613f25148ea6cc0
7fca037f9c14974f499195171bb84c38ebddc1e1
d4946b060a342cb215acca62a6c8ed379efefa75
3a4576e033bbed88541415e6a747c328cdd1f420
adbc9472741ebe00530a0fff14d0911d764801e4
23758e30bc5721b6a7e303caa4a8ae16c4e08c89
78c027fb36b9f42f330a0bc6f81ed422aa66b4ce
7d25bd50c711bfa50cce5d5bf8656c6e6cf903b1
8f6eab0f6c38d9e0245033b88331562d19164483
20baea401b309795261ec7bbed726acd2564c232
c9ac85a045b142e28ec9ae06da63b62c446152ab
9f731d185db45fbcda17a6524fc25c6deb2e4ef7
e012e62adea74d3fa54fdd27272d152adae75dce
2efeb05c6329d5cb5eabf8a62272dd17de7662ab
b0dbbb6b2a9518a2c83287c25881697e3afa256d
c20266164d76c6cf74c9337732555463b39721ba
db4529a1780141223899b4e79989e3a20ff640e4
39c3bacb7957b386aaea89df595638fe1fd452e1
c9030eb3cd021c4207126cd2371a921b8ab32691
5048c5efc74ed4f65f8e82518e25ca4d72974486
cee43f0de14020e91c047808f0ea092d92841c76
659b0e8189fdf99be678d5713259ba59726d991b
d5c297aa94d2a23163ad03280134547610e2076a
9928b6067da4c114f2be39a84e72552b382e60f1
e435a21715bc16a14eed156738154ddb8df12b84
399b8f986391d0bf90126814162a6eca0dae910a
d3d210eaee10aa6511e952b864015e43fbc9e098
8dacbbb2b25c145fe8806292acf3f340d7d6eb3d
94faf497ba246d6fd107d41ab8146cb02f979cf3
25ce2602078026b8065a64b35710d87d927f1b49
90e2a991f0660a215995e94ec4083b7b0f080753
2606ee33ad548bed1b9294bbca3962e834d12fd0
3c3cfd48fbc11fe94b87af613069fcb2dbf53cbd
6d417568ad68565fc7e01e7611ce8cf0b8fdb603
We cannot bisect more!
bisect run cannot continue any more

programmerjake, 2020-07-11:
so, it just skipped every commit between the good and bad commits -- I skipped if it didn't get far enough to output a vcd file with the signal included.

whitequark, 2020-07-11:
Okay, so bisect isn't useful here for some reason. I've tested general VCD writing with Minerva SoC and that works, so it's not completely broken.

programmerjake, 2020-07-13:
I think I found the cause of the bug:

In the broken code on line 348 it writes the current state to the vcd right *before* the next state is committed to the current state:
https://github.com/nmigen/nmigen/blob/30e2f91176edcd1c8766c2cb11f413b9c77936b9/nmigen/sim/pysim.py#L339-L354

In the working code it writes the current state right *after* the state is committed:
https://github.com/nmigen/nmigen/blob/303ea18cb60567e45a755c6b6289a601f27d46e6/nmigen/back/pysim.py#L245-L255

whitequark, 2020-07-13:
> I think I found the cause of the bug:

Thanks for the investigation! This is related to the way this code uses `pending` to speed up VCD writing (IIRC, writing all signals unconditionally took quite significant time). I'll fix this properly soon, but in the meantime, I believe this temporary fix should get your code working:

```diff
diff --git a/nmigen/sim/pysim.py b/nmigen/sim/pysim.py
index 8b0c054..c419cc4 100644
--- a/nmigen/sim/pysim.py
+++ b/nmigen/sim/pysim.py
@@ -255,7 +255,7 @@ class Simulator(SimulatorCore):
             for waveform_writer in self._waveform_writers:
                 for signal_state in self._state.pending:
                     waveform_writer.update(self._state.timeline.now,
-                        signal_state.signal, signal_state.curr)
+                        signal_state.signal, signal_state.next)
 
             # 2. commit: apply every queued signal change, waking up any waiting processes
             converged = self._state.commit()
```

programmerjake, 2020-07-13:
> Thanks for the investigation! This is related to the way this code uses `pending` to speed up VCD writing (IIRC, writing all signals unconditionally took quite significant time). I'll fix this [...]
we ended up making the same change in parallel :)

PR #430 (2020-07-13): write next state to vcd instead of curr state, since state is written just prior to commit
Fixes #429

I tested it and it seems to work!
@whitequark You can add a regression test later if you want.
commit c9fd0001 (2020-11-06): sim.pysim: avoid redundant VCD updates.

This commit properly addresses a bug introduced in 2efeb05c and then
temporarily fixed in 58f1d4bc.

Fixes #429.
closed 2020-11-06 by whitequark