amaranth-lang/amaranth #432: Add support for Display

whitequark, 2020-07-13:
A `Display` feature, similar to Verilog `$display`, has been requested a few times. We would depend on Yosys to implement it, which in fact is strongly in favor of having it.

Related to #427.

BracketMaster, 2020-11-01:
Why would yosys need to support the display cell?
Is the thought that there would be an 
``m.d.comb += Display(f’foo {bar}’}``?

And that yosys right now doesnt accept verilog with 
Display?

I don’t imagine that yosys has anything useful it could
do with Display except in the case of CXXrtl, so perhaps
Yosys would mostly ignore Display?

whitequark, 2020-11-01:
> Is the thought that there would be an `m.d.comb += Display(f’foo {bar}’}`?

Yes.

> I don’t imagine that yosys has anything useful it could do with Display

It should read and write Verilog with `$display` statements. Some reasons why that's necessary are:
  * LiteX integrates with nMigen through Verilog;
  * Using nMigen with iverilog or Verilator.

RobertBaruch, 2020-11-21:
I'm assuming we wouldn't use f-strings because that evaluates the expressions before passing control to the called function. Something more like `m.d.comb += Display(s, *args, **kwargs)` which does the equivalent of `s.format(args, kwargs)` except in `$display` language?

Also, is there an RTLIL equivalent?

whitequark, 2020-11-21:
> Something more like `m.d.comb += Display(s, *args, **kwargs)` which does the equivalent of `s.format(args, kwargs)` except in `$display` language?

Yep.

> Also, is there an RTLIL equivalent?

Not yet, but I'm working on it, like, right now.

RobertBaruch, 2020-11-21:
> Not yet, but I'm working on it, like, right now.

I kinda figured because now I recall the big meows about `$display` last week on Twitter :D

charlottia, 2023-08-22:
FWIW I have some WIP here that I'm planning to turn into something neat and PR-able in the coming days.

whitequark, 2024-03-04:
Support for this feature has been accepted via [RFC 50](https://amaranth-lang.org/rfcs/0050-print.html) and implementation is tracked in issue #1186.

closed 2024-03-04 by whitequark