amaranth-lang/amaranth #434: Use prebuilt yosys in CI

FFY00, 2020-07-13:
I realized the CI takes a lot of time to run because it builds yosys every time. Would you be interested in using a prebuild dev snapshot?

whitequark, 2020-07-13:
Yes, and in fact I've nearly finished working towards it, now that [YoWASP](https://yowasp.org) includes `yosys-smtbmc` and `sby`. But I'm open to other options.

whitequark, 2020-07-13:
(You might find it useful to know that `sby` is developed in lockstep with Yosys itself and it should in fact be installed not as a separate package but as a component of the feature complete Yosys package.)

whitequark, 2020-07-14:
> I realized the CI takes a lot of time to run because it builds yosys every time.

(Also, you probably know this, but it uses ccache so it doesn't build Yosys *every time from scratch*. It does rebuild large parts of it or the whole thing often enough that it's annoying, though.)

FFY00, 2020-07-14:
Well, I have a arch repo which rebuilds yosys daily, that would be my proposal. But as you develop yowasp I guess you'll probably go for it instead, as you have more control.

If you go with my arch repo, you can always blame things breaking on me :smile:

> (You might find it useful to know that `sby` is developed in lockstep with Yosys itself and it should in fact be installed not as a separate package but as a component of the feature complete Yosys package.)

From my packager POV I think it should be a different package. I can lock in dependencies, and these two are packages that should be really close together, you explained why. I am open to argue this though.

> (Also, you probably know this, but it uses ccache so it doesn't build Yosys _every time from scratch_. It does rebuild large parts of it or the whole thing often enough that it's annoying, though.)

Yeah. Just the linking on a weak cpu takes a long time :sweat_smile: and that would always be needed, even if ccache can provide every object.

whitequark, 2020-07-14:
> Well, I have a arch repo which rebuilds yosys daily, that would be my proposal.

Both Travis and GH Actions use Ubuntu or something, right?

> But as you develop yowasp I guess you'll probably go for it instead, as you have more control.

There's actually little difference for me, I just dislike Linux packaging in general, and since I already need YoWASP for other things, might as well do that. YoWASP can be quite slow, too.

> I can lock in dependencies, and these two are packages that should be really close together, you explained why. I am open to argue this though.

Well, the main argument I have is that sby (normally) installs into Yosys data directory. But if they're locked together then I don't really care one way or another. Maybe sby can recommend some SAT solvers and that's easier to do as a separate package.

FFY00, 2020-07-14:
> Both Travis and GH Actions use Ubuntu or something, right?

You can use a docker container.

Example: https://github.com/FFY00/dbus-objects/blob/master/.github/workflows/doc.yml

> I just dislike Linux packaging in general

I wonder if I could change your mind, I thought the same before arch. I've recently tried packaging for other distributions (mainly fedora) and I did not have a great experience, I could blame me on my lack of knowledge about the build system but arch had a much better experience getting started anyway. It's a pretty mild learning curve.

It's all bash, pretty straight forward. This is what it takes to package a stable yosys: https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/yosys

There is also a pretty great environment for reproducibility. I build everything in a clean chroot so that nothing leaks into the build.

> Well, the main argument I have is that sby (normally) installs into Yosys data directory. But if they're locked together then I don't really care one way or another. Maybe sby can recommend some SAT solvers and that's easier to do as a separate package.

Yeah, that isn't an issue for arch, it might be for other distros.

whitequark, 2020-07-14:
> You can use a docker container.

Please no :/

> I wonder if I could change your mind, I thought the same before arch.

Short answer: I'm used to Debian and too old to switch. Long answer: I should probably switch but the amount of effort required to do it is not something I can justify any time soon.

FFY00, 2020-07-14:
> Please no :/

It's natively supported, it is literally a 2 line configuration and the steps work just the same. Github caches the containers so the bring up time is pretty low.

> Short answer: I'm used to Debian and too old to switch. Long answer: I should probably switch but the amount of effort required to do it is not something I can justify any time soon.

Eh, I think it should be pretty straight forward, especially for someone like you. You'd essentially just `man pacman` to learn the package manager commands and that's it. There are no quirks in the distribution itself as you are one choosing how everything is set up. But anyway, I don't want to take more of your time. Feel free to ping me if you want to hear more :smile:.

whitequark, 2020-07-14:
> You'd essentially just `man pacman` to learn the package manager commands and that's it.

Yeah and that's exactly what I don't have time for. I don't really have stability in any other part of my life, let me have it at least in my damn choice of Linux distro.

commit 5ccc2122 (2020-07-16): CI: use WASM yosys instead of building our own.

Fixes #434.
closed 2020-07-16 by whitequark