amaranth-lang/amaranth #436: [RFC] Extending the linter interface to allow per-scope settings

whitequark, 2020-07-15:
Right now nMigen has an extremely simple linter that is also kind of hacky: it lets the first line of a file to contain a `# nmigen:` magic comment, and this comment allows disabling the `UnusedElaboratable` warning, mostly to allow creating elaboratables in tests just for their Python interface (i.e. without ever elaborating them). Issues like #404 would benefit from having a more well-defined linter interface.

This interface should allow:
  * enabling/disabling lints in file scope,
  * enabling/disabling lints in a scope delimited by the `with` statement,
  * avoid issues caused by globals in [Python's `warnings.catchwarnings`](https://docs.python.org/3/library/warnings.html#temporarily-suppressing-warnings).

Although not required, I think that it might be nice to copy Rust's `#[allow]`/`#[warn]`/`#[deny]` system, since it seems to work well for them, and I expect that nMigen users will request a way to make lints into hard errors in the future.

programmerjake, 2020-07-15:
seems like a good idea except that it is not the standard Python way.

programmerjake, 2020-07-15:
an additional concern is that there should be an API to disable linter features without requiring the python source file of the caller to be accessible, this is needed if using nmigen from some other programming language through CPython's C API.

whitequark, 2024-01-30:
While in principle still feasible, the direct motivation for this feature is gone now that #1059 obviates the need for configuring that particular warning.

I am going to close this issue now. This isn't to imply rejection of the proposal, but simply an indication that it should go through our new RFC process now (described at https://amaranth-lang.org/rfcs/).

closed 2024-01-30 by whitequark