amaranth-lang/amaranth #437: Utilise `write_verilog -sv` for targets that support it

Ravenslofty, 2020-07-16:
YosysHQ/yosys#2272 adds a SystemVerilog output option. It would be nice to use it on vendor targets that support SV, but would require a Yosys version check. It might be best to delay this until the next release of Yosys.

whitequark, 2021-12-11:
`write_verilog -sv` is included in the 0.10 release, required as of today, so this can be implemented now.

whitequark, 2024-02-09:
Broadly speaking this is still desirable, but not a high priority; anyone who is willing to do the work of finding out which of the vendor toolchains accept the Yosys SystemVerilog output and who feels strongly that SV should be used instead of the current Verilog output is welcome to send a PR.

closed 2024-02-09 by whitequark