amaranth-lang/amaranth #440: Wrong timestep with mixed clock edge ticks within pysim

kruemelkeks, 2020-07-19:
Take a module with a positive and a negative sensitive clock edge and excite it with pysim. When changing from `Tick('sync_neg')` to `Tick('sync')` the expected half cycle is added, but after a consecutive `Tick('sync')` the simulator also only runs another half cycle instead of a full one.

This behaviour can be observed since [2efeb05](https://github.com/nmigen/nmigen/commit/2efeb05c6329d5cb5eabf8a62272dd17de7662ab). 
An example reproducing the mentioned issue:
``` python
from nmigen import *
from nmigen.back.pysim import *

m = Module()
m.domains += ClockDomain('sync_neg', clk_edge='neg')
m.d.comb += ClockSignal('sync_neg').eq(ClockSignal('sync'))

a = Signal()
b = Signal()
m.d.sync += a.eq(1)
m.d.sync_neg += b.eq(1)

sim = Simulator(m)
sim.add_clock(1e-6, domain='sync')

def process():
    yield Tick('sync_neg') # <- ok
    yield Tick('sync') # <- ok
    yield Tick('sync') # <- wrong

sim.add_sync_process(process)
with sim.write_vcd("test.vcd", "test.gtkw", traces=[]):
    sim.run()

```

whitequark, 2020-07-22:
I believe this has been fixed in commit 58f1d4bcb67176bb91082ff3c2065fea9a10837e. Please check with the latest master.

closed 2020-07-22 by whitequark

whitequark, 2020-07-22:
I've also discovered another simulator bug based on your testcase. It is fixed in commit 1321c4591da619441dcea25c0bbfa3b224da3129.