amaranth-lang/amaranth #441: PS7 block not initialized on series-7 Zynq targets

DaKnig, 2020-07-19:
I'm using Vivado 2019.2, with the artix z7 board. Vivado spits this warning:
```
WARNING: [DRC ZPS7-1] PS7 block required: The PS7 cell must be used in this Zynq design in order to enable correct default configuration.
```
this should be simple enough to fix in verilog, but I don't know how to fix that in nMigen.

whitequark, 2020-07-20:
Try something like:
```
m.d.submodules += Instance("PS7")
```

whitequark, 2020-07-22:
Did this work for you?

DaKnig, 2020-07-23:
it did not.

whitequark, 2020-07-23:
Right, there's likely a bug in the RTLIL backend that prevents this from working. I'll take a look.

rroohhh, 2020-07-23:
From IRC:
```
[11:45:03] <vup> this seems to be the problematic line: https://github.com/nmigen/nmigen/blob/7aedb3e77045e865ea8e676b4a6616d03599a3ad/nmigen/back/rtlil.py#L858-L859 Considering Instances could be nested arbitrarily deep inside submodules without any ports this doesn't seem correct
```

whitequark, 2020-07-23:
@rroohhh Ah right, so the problem that 9faa1d37425ddafb5b2e76d502d86e3bff9ae54c was fixing is that Yosys treats empty modules as black boxes. Maybe they should contain a dummy wire or something.

whitequark, 2020-07-23:
Now that I think again about it, this might be a Yosys bug.

DaKnig, 2020-07-23:
from a question on the xilinx forums, it seems like putting a dummy with all the wires connected to some values (or disconnected- `open` in vhdl) should solve the issue. I am not sure this is optimal- this means many many lines of code for this one small problem.

commit 07a3685d (2020-08-26): back.rtlil: do not squash empty modules.

In commit 9faa1d37, the RTLIL backend was changed to ignore modules
without ports completely, since Yosys would recognize empty modules
as black boxes without explicit `write_verilog -noblackbox` and break
the design. That change had many flaws:
  * It removed instances without ports, which are used in e.g. SoC
    FPGAs to instantiate a dummy CPU.
  * It removed fragments without ports, which can appear in e.g. SoC
    FPGAs in case the fabric is not connected to any I/O ports.
  * Finally, it was just conceptually unjustified.

This commit changes th [... cut]
closed 2020-08-26 by whitequark