amaranth-lang/amaranth #442: Simulator with non-passive sync process and no clock added runs indefinitely without emitting a warning

The6P4C, 2020-07-21:
Take the following example, some basic synchronous logic and an even more basic sync process for the simulator:

```py
from nmigen import *
from nmigen.back.pysim import *

class Design(Elaboratable):
    def elaborate(self, platform):
        m = Module()

        s = Signal()
        m.d.sync += s.eq(1)

        return m

def proc():
    yield

sim = Simulator(Design())
sim.add_sync_process(proc)
# whoops, forgot to sim.add_clock(1e-9)
sim.run()
```

Since no clock is added to the simulator instance, the code never exits. It's a pretty easy line to miss and difficult to debug since it seems as if the simulator just... does nothing.

A warning when running the simulator with a sync process and no clock would be quite useful. A passive sync process obviously does not cause this - perhaps the warning should be raised on the first `yield` if the clock is undriven.

whitequark, 2020-07-21:
I don't think this is necessarily a bug. You could be driving a clock yourself, or you could be testing async logic, etc.

whitequark, 2020-07-21:
Oh wait, you mean warning if you are *waiting* on an undriven signal. We can do that, yeah.