amaranth-lang/amaranth #443: add API to get simulation time for debugging

programmerjake, 2020-07-22:
currently, `sim._state.timeline.now` must be used for pysim. It would be nice to have an official public API entrypoint for that instead.

whitequark, 2020-07-22:
Sounds reasonable.

programmerjake, 2020-07-22:
API suggestion:
```python
m = Module()
m.d.sync += Signal().eq(0)
sim = Simulator(m)

def process():
    yield Delay(12345e-6)
    now = sim.now # for use inside or outside processes
    print("current time:", now)
    now = (yield GetTime()) # for use inside processes -- useful when sim is not visible
    print("current time:", now)

sim.add_sync_process(process)
sim.add_clock(1e-6)
with sim.write_vcd("out.vcd", "out.gtkw", traces=[]):
    sim.run()
```

whitequark, 2020-11-07:
There's a good reason to not expose `sim._state.timeline.now` as it is: we shouldn't be representing time as floating point in the first place (I don't know what I was thinking when implementing that...) but we *really* shouldn't be exposing it to consumers.

Regarding `GetTime`: do you need to programmatically access the time, or do you just want to print it? In the latter case, would `yield Display("format...")` where the format language includes a substitution for the current time (like in Verilog) work for you?

whitequark, 2020-11-07:
I filed https://github.com/nmigen/nmigen/issues/535 to track the time representation issue.

programmerjake, 2020-11-08:
> Regarding `GetTime`: do you need to programmatically access the time, or do you just want to print it? In the latter case, would `yield Display("format...")` where the format language includes a substitution for the current time (like in Verilog) work for you?

We're currently just printing it, though it would be handy to have it (or a formatted version) available in case you want to have the output go somewhere other than `stdout`, e.g. json output for easy parsing.

whitequark, 2020-11-08:
> though it would be handy to have it (or a formatted version) available in case you want to have the output go somewhere other than `stdout`, e.g. json output for easy parsing.

I was thinking that Simulator would have an API to capture Display statements (as well as Assert, etc) from PySim as well as CXXSim.

whitequark, 2020-11-24:
Closing in favor of being able to print time via `Display` (including with custom formats), which will be shipped as a part of the `Display` feature itself, and being able to redirect `Display` output, which will be likely shipped later.

If new use cases arise we can reconsider this.

closed 2020-11-24 by whitequark