amaranth-lang/amaranth #453: submodules are allowed inside If blocks

jeanthom, 2020-07-29:
This code looks sketchy to me, but it's currently being allowed by nMigen:

```python
from nmigen import *
from nmigen.cli import main

class MyElab(Elaboratable):
	def __init__(self):
		...

	def elaborate(self, platform):
		m = Module()

		return m

class Top(Elaboratable):
	def __init__(self):
		...

	def elaborate(self, platform):
		m = Module()

		with m.If(1):
			m.submodules += MyElab()

		return m

if __name__ == "__main__":
	top = Top()
	main(top)
```

Console output:
```
sub-in-if.py:26: UnusedElaboratable: <__main__.Top object at 0x7f3f9bbcfd30> created but never used
```

whitequark, 2020-07-30:
I'm not sure how confusing it is. I'm hesitant to forbid it outright because right now nMigen doesn't really have abstraction mechanisms besides submodules (i.e. there is no other construct similar to a "function"). If you had an actual bug caused by this I would be more inclined to make it an error, but so far I'm not aware of anyone hitting it.

jeanthom, 2020-07-30:
I think it might be misleading to people who aren't familiar with HDL, thinking that they can add/remove elaboratables dynamically. By the way the generated RTLIL code looks correct (it will instantiate MyElab no matter the value of the condition).

whitequark, 2020-07-30:
A lot of the things would be misleading to people who aren't familiar with HDL, and if we removed all of them, we wouldn't have any HDL at all. There needs to be a stronger argument, like someone actually hitting the problem.

jeanthom, 2020-07-30:
No stronger argument ATM, I'm closing this issue.

closed 2020-07-30 by jeanthom