amaranth-lang/amaranth #455: cxxsim: synchronous assignment inside If statement is not executed

jfng, 2020-07-30:
(with the cxxsim branch checked out at 060ad2584edccad62e59ddd209f593a4dfc4670a)

Repro:
```python3
from nmigen import *
from nmigen.sim.cxxsim import *

m = Module()
i = Signal()
o = Signal()
x = Signal()

m.d.comb += x.eq(i)
with m.If(x):
    m.d.sync += o.eq(1)

def process():
    yield i.eq(1)
    yield Delay()
    yield
    yield Delay()
    assert (yield o) == 1

sim = Simulator(m)
sim.add_sync_process(process)
sim.add_clock(1e-6)
with sim.write_vcd("repro.vcd"):
    sim.run()
```

Output:
```
Traceback (most recent call last):
  File "repro.py", line 26, in <module>
    sim.run()
  File "/home/jf/src/nmigen/nmigen/sim/_core.py", line 202, in run
    while self.advance():
  File "/home/jf/src/nmigen/nmigen/sim/_core.py", line 191, in advance
    self._real_step()
  File "/home/jf/src/nmigen/nmigen/sim/cxxsim.py", line 124, in _real_step
    process.run()
  File "/home/jf/src/nmigen/nmigen/sim/_pycoro.py", line 121, in run
    self.coroutine.throw(exn)
  File "/home/jf/src/nmigen/nmigen/sim/_pycoro.py", line 62, in run
    command = self.coroutine.send(response)
  File "/home/jf/src/nmigen/nmigen/sim/_core.py", line 102, in wrapper
    yield from process()
  File "repro.py", line 21, in process
    assert (yield o) == 1
AssertionError
```

whitequark, 2020-07-30:
Better repro (no `Delay()`):

```python
def process():
    yield i.eq(1)
    yield
    yield
    assert (yield o) == 1
```

whitequark, 2020-08-27:
Okay, I figured out what happens here. There are two bugs. The first was finishing delta cycles too early, fixed in 7ca1477. The second one is that there's actually not enough information to trigger the process that's waiting on `Tick` because `clk` is single-buffered. This will require some rethinking of the cxxrtl interface.

whitequark, 2020-08-27:
I've pushed a low-performance workaround to the `cxxsim` branch. With this workaround I am able to run the Minerva "hello" example.

whitequark, 2020-09-03:
After the last refactor (pushed to `cxxsim` branch) I'm confident the root cause of this issue is no longer present.

closed 2020-09-03 by whitequark