amaranth-lang/amaranth #462: Pin direction and XDR can be set even if there is not matching subsignal

jeanthom, 2020-08-03:
```python
from nmigen import *
from nmigen_boards.ecpix5 import *

class Top(Elaboratable):
	def elaborate(self, platform):
		m = Module()

		ddr3 = platform.request("ddr3", 0, dir={"arrosoir":"-"},
			xdr={"chaise":4})

		return m

if __name__ == "__main__":
	top = Top()
	ECPIX585Platform().build(top)
```

The above code should at least trigger a warning.

whitequark, 2020-08-03:
IIRC it is done this way because some resources have optional pins, and otherwise you wouldn't have any way to request XDR on a pin if it exists, skipping it otherwise. I agree that this is not ideal.

whitequark, 2024-06-11:
Going forward the `xdr=` parameter and non-`dir-"-"` `request()` calls will be deprecated so this doesn't seem important to address any more.

closed 2024-06-11 by whitequark