amaranth-lang/amaranth #478: Support division of negative numbers?

pepijndevos, 2020-08-17:
It appears that Yosys now supports `$divfloor` as of https://github.com/YosysHQ/yosys/commit/edd8ff2c0778d97808869488cc7394151456c4ca

In `ast.py` it is mentioned that the different semantics mean it's not supported:
https://github.com/nmigen/nmigen/blob/e46118dac0df315694b0fc6b9367d285a8fc12dd/nmigen/hdl/ast.py#L173-L179

I'm not sure what division synthesizes to in Yosys in general, or if it supports the general case at all, but it seems that on the nMigen side we can now emit `$divfloor` and fully support division with correct semantics. I'd be happy to put together a PR for this at some point.

whitequark, 2020-08-17:
Reasonable. I think we might want to wait until there's a Yosys release, because we can't shim over the lack of `$divfloor` in older Yosys in `back.verilog`, like we do for other Yosys-related variability.

pepijndevos, 2020-08-17:
The verilog backend in that commit (which I do not claim to fully understand) does seem to "shim" over the lack of flooring division in Verilog, so I think it might technically be possible to work around it, but I agree waiting for a Yosys release is easier. Bit unfortunate that Yosys releases are not exactly frequent.

whitequark, 2020-08-17:
> I think it might technically be possible to work around it

I guess you could, when targeting older Yosys, always provide a `$divfloor` cell as a part of `nmigen.back.verilog`, and then immediately flattening it, so that it never actually appears in output Verilog. But it's not clear that the effort is justified.

pepijndevos, 2020-08-17:
Probably not...

whitequark, 2023-02-03:
This is implemented as of b452e0e8 (issue #621).

closed 2023-02-03 by whitequark