amaranth-lang/amaranth #507: How to implement support for arrays on ports

fl4shk, 2020-10-18:
So, Verilog doesn't support arrays on ports, only vectors.  The best way to get around this limitation in Verilog, in my opinion, is to encode the array into a vector.  This works even for `Array`s of `Record`s.

I suggest perhaps making a `PackedArray` class (like in SV), derived from `Value`, that takes a shape and a size.  I believe this would also solve the issue of not being able to place an `Array` inside of a `Record`.

I'll work on developing an example of it.

fl4shk, 2020-10-18:
I have decided to wait for `ValueCastable` to be merged before I build this.

whitequark, 2020-10-18:
I don't think that any changes related to `Record`s should be done before the `Record` redesign (#342). In general, I would prefer to enable you to use `ValueCastable` to do this change localized in your codebase than to add it to the core language.

fl4shk, 2020-10-18:
Sounds good to me.

whitequark, 2020-10-22:
I believe this question got resolved on IRC, so closing.

closed 2020-10-22 by whitequark