amaranth-lang/amaranth #510: [RFC] Allow DiffPairs and Pins to be inverted individually

anuejn, 2020-10-20:
Currently DiffPairs and Pins cann be either inverted completely or not at all.
This is a problem in corner cases where the hardware inverts some parts of a logical Bundle of ios (for example to ease routing in case of differential pairs; see #386). 

I herby propose to extend the Platform DSL in the following way:
* Add a `invert=` kwarg to the constructor of `DiffPairs`
* Allow Tuples of the width of the pin list to be passed to the `invert=` kwarg of both `DiffPairs` and `Pins`

whitequark, 2020-10-20:
Right, this is the obvious solution. The main problem I have is that when you have a lot of diffpairs, it can be hard to correlate the booleans in the `invert=` tuple with the actual diffpairs.

whitequark, 2020-10-20:
As an alternative I was wondering if something like `DiffPairs("A1 !A2", "B1 !B2")` would work. The `!` sigils would then have to match in both the P and the N pin lists.

adamgreig, 2020-10-20:
I like putting it in the pin list string(s), it's compact but I think also quite clear. Maybe "~A1" would be a better match to the nmigen negation syntax?

whitequark, 2020-10-20:
Sure, `~` is fine by me.

anuejn, 2020-10-20:
I also really like that idea :)

whitequark, 2020-10-20:
Sounds great. I'll bring this up on the next meeting but I think you can start implementing it as it seems unlikely anyone's going to be opposed.

anuejn, 2020-10-20:
sounds good :)

anuejn, 2020-10-21:
I think we should support similar syntax for the use with connectors (which can be routed through e.g. inverted level shifters or have crossed differential pairs). So I think it would be nice to have syntax elements in connectors that can say "invert this signal if it is a diff pair" or "invert this signal always".

whitequark, 2020-10-21:
I would like to avoid introducing that functionality (and the associated syntactic burden) until someone actually needs it.

anuejn, 2020-10-21:
/me needs the latter (inverted differential pairs in connectors)

whitequark, 2020-10-21:
Okay. I still think it should be done separately because while this one is fairly straightforward (it's just a way to write a tuple that's less error prone than usual), what you request with connectors requires completely rethinking the way we do connectors.

anuejn, 2020-10-21:
Okay I see that.
So I will implement this RFC as-is and go with out of tree hacks for now.
If I really need this feature to become upstream I will write another RFC :)

whitequark, 2020-10-21:
I think we can do it upstream; I said [earlier](https://github.com/nmigen/nmigen/issues/510#issuecomment-713453939) that I'd like to avoid introducing it because it sounded theoretical (which is usually bad because features done in abstract tend to not survive collision with reality). If you actually hit that problem I think we can certainly add it. Let's discuss the connector semantics on the next meeting?

anuejn, 2020-10-21:
Okay :+1:

whitequark, 2020-10-26:
We discussed inversion of connector signals at the last meeting. Because inversion of single ended signals is rarely needed and can also complicate things even further (you have to keep track of two layers of inversion rather than one), we decided that we would implement inversion in connectors but only for diff pairs. Connecting a single-ended signal to an inverted connector pin would be a hard error. If it turns out to be worth the complexity in the future we can always accept it.