amaranth-lang/amaranth #515: Provide public api to add build commands

anuejn, 2020-10-26:
It would be nice to have (semi) public API to inject custom build commands into the build_name.{sh,bat} files.
My specific use case is to generate a bash script that contains a base64 encoded version of the bit stream that loads the bit stream automatically via the linux fpgamanager infrastructrure  on a zynq. 
Currently, I am having [quite a hack in place](https://github.com/apertus-open-source-cinema/nmigen-gateware/blob/master/src/soc/soc_platform.py#L31) to implement that but that constantly breaks and is a very bad solution.

For me live would be easier if `TemplatedPlatform` had an extra_command_templates property that is guaranteed to be not a property object and therefore easy to override.

whitequark, 2020-10-26:
> My specific use case is to generate a bash script that contains a base64 encoded version of the bit stream that loads the bit stream automatically via the linux fpgamanager infrastructrure on a zynq.

This sounds like it should be a part of an external programmer. Could you explain that workflow in a bit more detail?

anuejn, 2020-10-26:
At first that logic was inside the `toolchain_program()` function of the platform but I realized that I want that bash script to be generated for later use even if I do not program the FPGA directly. I see that file a bit more like what a svf file is compared to the normal bitstream format.

whitequark, 2020-10-26:
Ack, that makes sense. I'll see what can be done.

anuejn, 2020-10-26:
My workflow involves generating that file (which also contains other stuff from the design like a memorymap).
This file is then copied over to the zynq via scp and executed via ssh in the `toolchain_program()` function.

anuejn, 2021-06-08:
see #618. Not needed.

closed 2021-06-08 by anuejn