amaranth-lang/amaranth #519: Xilinx Zynq: generated bitstreams do not work with (recent?) FPGA manager

nfbraun, 2020-11-01:
On Xilinx Zynq, the task of loading bitstreams into the programmable logic part
of the device is typically handled by the Linux kernel, via the FPGA Manager
framework.

The bitstreams that nMigen generates (via Vivado) fail to load through FPGA
manager, with an error message like this:

    fpga_manager fpga0: writing zynq_image.bin to Xilinx Zynq FPGA Manager
    fpga_manager fpga0: Invalid bitstream, could not find a sync word. Bitstream must be a byte swapped .bin file
    fpga_manager fpga0: Error preparing FPGA for writing

As the error message says, that is due to the fact that mainline FPGA manager
expects a byte swapped .bin file, but the file produced by Vivados
`write_bitstream -bin_file` command is not byte swapped. The (obsolete) xdevcfg
interface in ancient Xilinx kernels seems to have detected this (via the sync
word) and corrected it if required, but this feature does not seem to have made
it into FPGA manager.  (See https://lore.kernel.org/patchwork/patch/605909/ for
the discussion.)

There is a tool called `fpga-bit-to-bin.py` floating around on the Internet
that people use to generate the correct .bin files; however, it can also be
done directly from Vivado.

My suggested fix would be to not generate the .bin file via the `-bin_file`
option to `write_bitstream` (in `nmigen/vendor/xilinx_7series.py`), but instead
issue a separate command to generate it,

    write_cfgmem -force -format bin -interface smapx32 -disablebitswap -loadbit "up 0 {{name}}.bit" {{name}}.bin

This would give a .bin file that works with FPGA manager.

If there are uses that require a non byte swapped .bin file, one could also
generate two different .bin files, but I don't think that is required (the
xdevcfg interface works with both variants).

If this solution is deemed acceptable, I can submit a PR.

whitequark, 2020-11-02:
Your solution seems good to me.

closed 2020-11-03 by whitequark

TanviWani, 2025-02-25:
when i write "cat /root/{{name}}.bin > /dev/xdevcfg nothing happens to the output or terminal and "echo {{name}}.bin > /sys/class/fpga_manager/fpga0/firmware" gibes invalid argument

PR #523 (2020-11-03): vendor.xilinx_7series: byte swap generated bitstream
The Zynq driver in the FPGA Manager framework on Linux expects bitstreams that
are byte swapped with respect to what the Vivado command
`write_bitstream -bin_file` produces. Thus, use the `write_cfgmem` command with
appropriate options to generate the bitstream (.bin file).

Fixes #519.
commit 14a5c42a (2020-11-03): vendor.xilinx_7series: byte swap generated bitstream

The Zynq driver in the FPGA Manager framework on Linux expects bitstreams that
are byte swapped with respect to what the Vivado command
`write_bitstream -bin_file` produces. Thus, use the `write_cfgmem` command with
appropriate options to generate the bitstream (.bin file).

Fixes #519.