amaranth-lang/amaranth #531: Implement missing CXXRTL features

whitequark, 2020-11-06:
Integrating CXXSim (#324) requires a number of improvements to the Yosys CXXRTL backend that have to be done upstream. While #324 is nMigen/CXXSim-focused, this issue is Yosys/CXXRTL-focused. The underlying tasks it solves are just those in #324 though.

## Remaining tasks

### Blockers (to be completed pre-merge)
 - [ ] CXXRTL generated code should have an option to disable assertions of RTL invariants without enabling `-DNDEBUG`, since `-DNDEBUG` would also disable assertions of CXXRTL API contracts, and most if not all CXXRTL embedders do not want the generated code to `abort()`.
 - [ ] CXXRTL designs should have a dedicated `reset` method. Although it is always possible to destroy and recreate the entire simulation, this has two downsides:
    * Memory addresses in debug items will get invalidated and will need to be reacquired. This may be quite costly to Python code.
    * The issue above can be partially solved by assigning a new default-initialized instance on C++ side using a new C API function. However, this would still destroy the state of every black box. Given that black boxes are explicitly intended to acquire heavyweight resources like GUI windows and OS network interfaces, this is undesirable.
 - [ ] Investigate whether memory writes can be done using the C API without race conditions.

### Non-blockers (may be completed post-merge)
 - [ ] Add support for Yosys enumerated signal attributes and/or arbitrary user-specified decoders to the CXXRTL VCD writer.
 - [ ] Add basic support for `$write` and `$assert` (& its related cells) that translate to console output and RTL assertions respectively.
 - [ ] Add advanced support for `$write` and `$assert` (& its related cells) that queue these events for consumption through the C API. (#432, #427)
 - [ ] Investigate whether some kind of shared event manager/dispatcher would be a good solution to systematic issues with edge-triggered logic. (#439)

whitequark, 2020-12-03:
Most of the blockers are done. Only the memory write issue remains.

whitequark, 2020-12-08:
An IRC discussion with @awygle confirmed my suspicion that race-free concurrent operations on memories are not something that has a lot of value, and given the significant amount of effort and complexity required to support them, it makes sense to not provide them at all.

As a consequence, memory reads or writes from testbenches can only be, in general, safely done while the design is held in reset. In some specific cases, concurrent reads/writes from a testbench are possible while the design is running, but requires case-by-case evaluation for hazards.

mithro, 2020-12-08:
It would be nice if there was a way to detect if concurrent operations on memories are occurring and force an error (maybe a `-Wconcurrent-memory -Werror` style thing?).

whitequark, 2020-12-08:
The thought of a "RTL TSAN" crossed my mind, but I do not expect to have time to work on it in foreseeable future.