amaranth-lang/amaranth #542: Using ClockSignal in platform.add_clock_constraint

rroohhh, 2020-11-11:
Currently one cannot use a `ClockSignal` in `platform.add_clock_constraint`:
```python3
from nmigen import *
from nmigen_boards.microzed_z010 import *

plat = MicroZedZ010Platform()
plat.add_clock_constraint(ClockSignal(), 25e6)
```
fails with
```
Traceback (most recent call last):
  File "a.py", line 5, in <module>
    plat.add_clock_constraint(ClockSignal(), 25e6)
  File "/home/robin/.guix-profile/lib/python3.8/site-packages/nmigen/vendor/xilinx_7series.py", line 184, in add_clock_constraint
    super().add_clock_constraint(clock, frequency)
  File "/home/robin/.guix-profile/lib/python3.8/site-packages/nmigen/build/res.py", line 223, in add_clock_constraint
    raise TypeError("Object {!r} is not a Signal".format(clock))
TypeError: Object (clk sync) is not a Signal
```
It be nice to be able to use a `ClockSignal` directly in `platform.add_clock_constraint`.

whitequark, 2020-11-11:
That's not possible--the `Platform` does not know the context of any particular `ClockSignal`, and without the context, it is ambiguous.

awygle, 2020-11-12:
I'd like to recommend enhancing the error message for this case, or else explaining in more detail here for future reference. Even with your comment above, it's not immediately obvious to me why `plat.add_clock_constraint(ClockSignal(), 25e6)` doesn't work but

```python
cs = ClockSignal()
s = Signal()
m.d.comb += s.eq(cs)
plat.add_clock_constraint(s, 25e6)
```

does.

awygle, 2021-02-10:
Throwing some spaghetti at the wall for the error message:

`TypeError: Object (clk sync) is not a Signal, it is a ClockSignal. You may be able to assign the value of ClockSignal to a Signal, which can then be used to add a clock constraint.`

Thoughts?

PR #1288 (2024-04-07): build.res: give a more specific error for `add_clock_constraint(ClockSignal)`
Fixes #542.
commit 38ad3575 (2024-04-07): build.res: give a more specific error for `add_clock_constraint(ClockSignal)`.

Fixes #542.
closed 2024-04-07 by whitequark